ShikOfTheRa/scarab-osd #478: OSD display problem (with telemetry switch = ON)

tfoutfou, 2018-02-28:
Hardware : Spracing f3 -- Frsky X4R SB -- Micro Minim OSD -- TBS VTX with smartaudio

BF : 3.2.5 OSD : 1.7.4.0

UART1 : USB shared with smartport telemetry (on a switch with option tlm_switch = ON )
UART2 : MSP connected to OSD
UART3 : X4RSB
softserial 1 : TBS Smartaudio

Hi , with the hardware and configuration above .
When i enable the telemetry switch on a RC dedicated channel i get an OSD problem in the MODE display.
Instead of having : [STAB] [HOZN] [ACRO] when i enable the telemetry switch the MODE change to [WPT] : 0 (which is not logic at all since it as nothing to do with any mode whatsoever )

So i think their is a problem somewhere in betaflight OR scarabosd .
Could you please tell me if it come from ScarabOsd and maybe fix it. Thank you.

(i already post the issue on betaflight repo and got this answer)

> Looks like scarab OSD reads modes by index. Betaflight has switched to using permanent ids for modes some time ago, which has the advantage of the ids not changing when new modes are added / old modes deleted.
scarab OSD should switch to reading permanent mode ids from Betaflight. Please open a ticked with scarab OSD for this.

>

ShikOfTheRa, 2018-02-28:
OK - I will have a look at this. I wasn't aware Betaflight had changed its operation.

ShikOfTheRa, 2018-02-28:
Hi - I have added support for this. It is in latest repository, 

This is now on R1.8 beta test ready for release. I have just uploaded so can do easy install from GUI

See post here for details how to get:
http://fpvlab.com/forums/showthread.php?34250-MWOSD-for-BETAFLIGHT-CLEANFLIGHT-RACEFLIGHT-PIXHAWK-KISS-DRONIN-INAV-and-others&p=844916&viewfull=1#post844916

tfoutfou, 2018-02-28:
i have another problem that as nothing to do with this isssue , and i dont know how to be in direct contact with you .

I have bought some time ago the chrome store version of the configurator and so i dont have a registration key for standalone configurator , what can i do about that ?
because right now i can t use 1.8 because of this .
thank you

ShikOfTheRa, 2018-02-28:
Hi - can you send me the order number from chrome store so I can get a key generated. Send to: 

gary (@)
flypix.  co. uk

tfoutfou, 2018-03-01:
It seem to be OK in the last release , thank you

closed 2018-03-01 by tfoutfou