ShikOfTheRa/scarab-osd #481: GUI : layout option switch for "Glidescope"

tfoutfou, 2018-03-01:
Hi, 
In the last GUI 1.8.0.0 , when i launch the program and DO NOT connect i have the Glidescope switch available .
As soon as i connect to my FC (passtrough FIY) , the Glidescope switch disapear , and so i cant turn it off for a layout.
(I triple check this by counting the number of switch , it is the only one vanishing)
The workaround is to load a layout where glidescope is already disabled but it is still a little issue.

(same configuration as in the previous issue)

Thanks

ShikOfTheRa, 2018-03-01:
Hi, I think that is small bug between operation of GUI and OSD
on OSD glidescope is enabled only if the aircraft type is fixed wing. 
GUI disables switch if sees fixed wing.
But as you say if its already enabled you can't disable and it shows on preview.

Will get this resolved - thanks had not noticed this.

ShikOfTheRa, 2018-03-02:
Confirmed and resolved for the next beta test release - or full 1.8 release

closed 2018-03-02 by ShikOfTheRa