ShikOfTheRa/scarab-osd #484: ublox module configuration

nkalgin, 2018-03-05:
Good day.
Could you please specify. which UBX commands have to be configured in ublox module(M8N as example) for correct connection between ublox and OSD?

nkalgin, 2018-03-05:
I was trying to connect ublox m8n drectly to minim micro OSD(directly through instruction), but i've seen that sattelites indicator constantly swithed between real value(for example 12 sattelites) and 0, and both coordinates stating blank(zeros).
So I suspect, that some UBX messages not correctly configured on M8N device.

ShikOfTheRa, 2018-03-05:
If RX/TX connected it should self configure GPS at boot. I have never used ucenter or anything to set up.

More or less everything worked out from following::
    case MSG_POSLLH:
        GPS_fix 
       GPS_coord[LON
        GPS_coord[LAT
        GPS_altitude
        GPS_altitude_vario 
    case MSG_SOL:
      GPS_numSat 
      GPS_dop 
    case MSG_VELNED:
      GPS_speed 
      GPS_ground_course 

These are commands:

 0xB5, 0x62, 0x06, 0x01, 0x03, 0x00, 0xF0, 0x05, 0x00, 0xFF, 0x19,                  //disable all default NMEA messages
  0xB5, 0x62, 0x06, 0x01, 0x03, 0x00, 0xF0, 0x03, 0x00, 0xFD, 0x15,
  0xB5, 0x62, 0x06, 0x01, 0x03, 0x00, 0xF0, 0x01, 0x00, 0xFB, 0x11,
  0xB5, 0x62, 0x06, 0x01, 0x03, 0x00, 0xF0, 0x00, 0x00, 0xFA, 0x0F,
  0xB5, 0x62, 0x06, 0x01, 0x03, 0x00, 0xF0, 0x02, 0x00, 0xFC, 0x13,
  0xB5, 0x62, 0x06, 0x01, 0x03, 0x00, 0xF0, 0x04, 0x00, 0xFE, 0x17,
  0xB5, 0x62, 0x06, 0x01, 0x03, 0x00, 0x01, 0x02, 0x01, 0x0E, 0x47,                  //set POSLLH MSG rate
  0xB5, 0x62, 0x06, 0x01, 0x03, 0x00, 0x01, 0x03, 0x01, 0x0F, 0x49,                  //set STATUS MSG rate
  0xB5, 0x62, 0x06, 0x01, 0x03, 0x00, 0x01, 0x06, 0x01, 0x12, 0x4F,                  //set SOL MSG rate
  0xB5, 0x62, 0x06, 0x01, 0x03, 0x00, 0x01, 0x12, 0x01, 0x1E, 0x67,                  //set VELNED MSG rate
  0xB5, 0x62, 0x06, 0x16, 0x08, 0x00, 0x03, 0x07, 0x03, 0x00, 0x51, 0x08, 0x00, 0x00, 0x8A, 0x41, //set WAAS to EGNOS
  0xB5, 0x62, 0x06, 0x08, 0x06, 0x00, 0xC8, 0x00, 0x01, 0x00, 0x01, 0x00, 0xDE, 0x6A //set rate to 5Hz

ShikOfTheRa, 2018-03-08:
Hve you had any luck?

nkalgin, 2018-03-09:
It strongly help, thanks. But the rason was in other case: i'm using 2 RF UART modems between OSD and Ublox M8N module, but they are only half-rate... so in start GPS module start sending standart NMEA pach of commands with GPS state, and comleatelly blocks any attempts of recieving commands from another side. So OSD just can't send anything to reconfigure modele dor UBX.
I hope, that full-rate modems would solve the problem.

Anyway, thank you very much for advance.

closed 2018-03-09 by ShikOfTheRa