ShikOfTheRa/scarab-osd #497: need clarifaication GUI pc app / chrome app

macfly1202, 2018-03-13:
In this morning mail about mwosd 1.8 release, you state that user must buy 1.8 PC app on the site https://www.virtualpilot.co.uk 
I bought the only OSD on this site and I got chrome 1.7.19 app.... The app I already own... 

Could you clarify where is the 1.8 bought PC app ? It seems there is a dev app 1.8  here https://github.com/ShikOfTheRa/scarab-osd/tree/master/MW_OSD_GUI/R1.8 

Thanks for clarification and the big job on PX4 firmware since weeks ...

ShikOfTheRa, 2018-03-14:
Hi,

It is a little confusing as we are releasing a new version and having to transfer chrome users to a PC app. I will try to clarify! :

On chrome store the version is 1.7.1.9 which supports 1.7 MWOSD

As google has announced closure of the chrome store we started to produce 1.7 and 1.8 versions of PC app ready to migrate users. 1.8 moved out of development yesterday.

95% of users are on Chrome – and the migration started yesterday.

Users will start to see their apps upgrade to 1.7.1.10 – this provides a key and instructions for enabling the transition to PC apps (and 1.8)

The good news… !
There is no cost to transfer or need to buy again 1.8.

If can’t wait for the google upgrade (I have no control over who and not sure how long it will take…), drop me an email to gary (at) flypix.co.uk so I can request key to be generated / sent.

macfly1202, 2018-03-16:
thanks. Clarification ok. Mail send. Close Issue

closed 2018-03-16 by macfly1202