ShikOfTheRa/scarab-osd #498: no data message

retif, 2018-03-24:
i have 1.8 mwosd minimosd inav plane firmware and 1.8.1 inav, connecting minimosd through softserial. and it show me message "no data". Previously i used other firmware it mostly worked, but had some glitches, so i decided to move, but it not works at all for me, i think the problem may be in port speed which is limited to 19200 for softserial, but how can i change port speed on minimosd mwosd?

ShikOfTheRa, 2018-03-26:
Hi, port speed is fixed in pre-compiled GUI frimware
 - you need to compile and upload your own version.  

here is guide. Look in config.h for baud rate. Once changed, don't forget to change GUI baud rate to be able to talk to OSD at new rate. 
https://github.com/ShikOfTheRa/scarab-osd/wiki/Custom-installation-guide

retif, 2018-03-26:
thanks, it worked

closed 2018-03-26 by retif