ShikOfTheRa/scarab-osd #499: Timeout while trying to flash Furious FPV Innova

andrejpodzimek, 2018-04-02:
I'm updating a [Loki X3S](http://www.helipal.com/download/pdf/helipal-srd130S-BetaFlight-v1.pdf) with a `PIKOBLX` board (running Betaflight 3.3, `FF_PIKOBLX` target). It appears to have [Furious FPV Innova](http://furiousfpv.com/product_info.php?products_id=76) OSD/VTX.

Connecting to the VTX from the 1.8 GUI works fine, both using `serialpassthrough 1 115200` in Betaflight or using the `UART2` option in the GUI. Reading finishes, I can see voltage/current/rssi numbers etc. and I get a _version mismatch warning_. (That's because the OSD has something older than 1.7 flashed.)

Sadly, flashing doesn't work. The only message I see in the flashing menu is _Waiting for bootloader reset_ and right afterwards I get ***Error: programing FAILED: Sending 3020: receiveData timeout after 400ms***. Options I selected in the menu:

* Version 1.8.0.0
* Device minim
* Variant betaflight
* Type multirotor

I also tried microminim, but got the same error message. (At which point I stopped experimenting, because I guess flashing the wrong variant may just brick it.)

Flashing doesn't work in the older (Chrome-based) GUI 1.7 either. It produces the same error message. But reading and configuration works there, without version mismatch warnings.

Is there a way to update this OSD to 1.8 or to 1.7.4.1? Does it need some special steps to get into the OSD mode?

andrejpodzimek, 2018-04-02:
Oh, I've just noticed #327. That doesn't look good. :sob: The Banggood clones also don't seem to be available any more. Well, maybe I should ditch the whole 2x2 cm stack from the Loki and replace it with [something more recent](https://www.banggood.com/HGLRC-XJB-F440-TX20_V2-Omnibus-F4-Flight-Controller-OSD-40A-Blheli_32-3-5S-ESC-25100200350mW-VTX-p-1242937.html).

ShikOfTheRa, 2018-04-03:
Flashing through passthrough is unlikely to work - can you go directly to the OSD/VTX?

andrejpodzimek, 2018-04-03:
So I assume I'll need to solder it to a suitable usbserial device. Would [this](https://www.banggood.com/FTDI-5V-Micro-USB-Turn-To-TTL-MWC-CP2104-Debugger-Programming-Unit-p-1181312.html) or [this](https://www.banggood.com/6-In-1-CP2102-USB-To-TTL-485-232-Converter-3_3V-5V-Compatible-Six-Multifunction-Serial-Module-p-1200130.html) work?

Is there a suitable device type in the default GUI or should I just build a custom mwosd firmware image with [this line](https://github.com/ShikOfTheRa/scarab-osd/blob/master/MW_OSD/Config.h#L18) uncommented?

ShikOfTheRa, 2018-04-04:
Both those look OK.
There isn't a device currently in the GUI - mainly because I haven't used or tested myself. I believe it does from the feedback in #327 - but happy to request GUI refund if it doesn't

closed 2018-05-21 by ShikOfTheRa