ShikOfTheRa/scarab-osd #500: Missing documentation on mwosd device types

andrejpodzimek, 2018-04-02:
I'm trying to flash 1.8 on [this device](https://www.banggood.com/MXK-Power-Distribution-Board-PDB-with-OSD-BEC-Output-5V-12V-3A-LC-Filter-for-CC3C-Naze32-F3-p-1031375.html). Selecting `minimosd` yields weird votage / current / other readings that normally didn't show in the old 1.6 version when the device wasn't powered (from sources other than USB). So I'm assuming that I selected the wrong device, but don't want to go on a trial-and-error spree to avoid bricking it completely.

How can one identify which device should be selected? I couldn't find any detailed log messages in the mwosd GUI or other hints. The documentation doesn't say anything about device types, how they differ, how to identify them etc.

FTR, here's my dmesg output when I connect the device:

    [63004.406469] usb 3-2.4: new full-speed USB device number 33 using xhci_hcd
    [63004.496134] usb 3-2.4: New USB device found, idVendor=10c4, idProduct=ea60
    [63004.496140] usb 3-2.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
    [63004.496145] usb 3-2.4: Product: CP2104 USB to UART Bridge Controller
    [63004.496149] usb 3-2.4: Manufacturer: Silicon Labs
    [63004.496152] usb 3-2.4: SerialNumber: 012074DA
    [63004.497079] cp210x 3-2.4:1.0: cp210x converter detected
    [63004.498221] usb 3-2.4: cp210x converter now attached to ttyUSB0

andrejpodzimek, 2018-04-02:
[This mentions `minimosd`](http://en.eurny-rc.fr/review-mxk-power-distribution-board-pdb-with-osd-bec/), so it's probably `minimosd`, indeed. These readings look surprising though (and used to be 0 or much smaller numbers with older mwosd while on USB only):

[image]

ShikOfTheRa, 2018-04-03:
I hadn't seen that before.

So minimosd would be correct.
Volts/amps/rssi etc will possibly be random as it seems they are not connected on the hardware. Must select  "Use FlightController" for vbat, , rssi and current
It looks same as original minimosd - can only take volts etc. from

ShikOfTheRa, 2018-04-09:
Hope that naswers question - in brief readings are invalid for that OSD hardware

closed 2018-04-09 by ShikOfTheRa