ShikOfTheRa/scarab-osd #502: Implement MWOSD in custom board

hexagon-robotics, 2018-04-03:
Hello,

I want to implement MWOSD in my board, I have seen someones .ino files. Is it only possible to implement it in Arduino-Atmel mcus? I use other mcu.

Is there a link that deals with this, or a tutorial to implement it?

Best regards,
Javier

hexagon-robotics, 2018-04-03:
Sorry, I found this --> https://github.com/ShikOfTheRa/scarab-osd/wiki/Quick-start-guide

hexagon-robotics, 2018-04-03:
About Arduino bootloader, it's possible to port scarab-osd to another bootloader?

ShikOfTheRa, 2018-04-04:
Hi - Yes it is atmel mcu only today
Probably other bootloader are OK - but note the GUI is designed to write firmware to default arduino code locations.

can email me at gary <at> flypix.co.uk if you would like to discuss further. Probaly easier via email

hexagon-robotics, 2018-04-04:
Thank you very much!

I write you soon.

I use Oracle JavaFX for the GUI's, for example to SKYFX, the multiplatform configurator app to the flight controller.  I could to include in SKYFX or in a standalone application, if porting is not too complicated, I would have to see it.

Best regards

closed 2018-04-04 by hexagon-robotics