ShikOfTheRa/scarab-osd #503: [Question] display betaflight rtc on summary

fape, 2018-04-05:
Do you think it's worth to implement rtc (date/time) displaying on summary screen?
If the pilot uses taranis + lua scripts the date and time on the fc will be updated from the taranis rtc (unfortunately but there is some condition :S). And we can query via MSP (MSP_RTC). Eg every time the summary screen is displaying we query the actual date and time and displaying on the screen.
I hope it can be useful for sorting dvr footage.
What do you think?
I think I can implement it ;)

ShikOfTheRa, 2018-04-09:
oops - thought I answered this.

Actually I think some would be interested it as a display option on screen layout as well. I was thinking more from those with GPS FC or GPS direct.

As it's quite specific / very low on memory / not been requested previously  I would want it to be an option in config. h and disabled by default. Mainly memory...

ShikOfTheRa, 2018-05-21:
Hi Fape - I have added this to dev roadmap so will close this thread. Trying to keep the issues list low for outstanding bugs etc. 
I would like to have this added for next release.

I will update here so you get notification as well even when closed..

fape, 2018-05-21:
Thanks. I think I can find sometime time for that feature very soon. Hopefuly in this week. 
(My daytime job is software developer [what a surprise :D] and we had a very big release last week. So I was very busy in last few weeks.)

ShikOfTheRa, 2018-05-22:
Fape - re-opening discussion as I had a little think about this last night. Think it would be quick for me to get it roughly working as already most code is in there as it was added years ago for a multiwii extension. 
Bear in mind there is very little memory on the chip so might have to take some tough decisions:

1 Display only at armed / initial display menu or at any time.
Displaying raw UTC date time would be easiest - no corrections for time zone.
Compensating for Summers time - another complexity
Both are currently supported and in the OSD menu for manual adjustment

because memory is so tight I am leaning towards:

config option 1 - default
A low memory UTC only approach for default
config option 2 - OSD menu adjust with TZ and Summer time support. 

GUI support will also have to be provided to enable and move around on display. Its there now - just hidden.

Thoughts?

ShikOfTheRa, 2018-05-25:
OK - have this in the master.
It is UNTESTED at moment and will require GPSTIME and MSP_RTC_SUPPORT enabling.

It is waiting for me to get a FC/GPS setup that I can test it with. Should be early next week.
Most likely will not work and need some debugging :)

ShikOfTheRa, 2018-06-01:
Good news:
It works with iNav now.
Bad news:
I just looked at BF - its supported in a completely different way. grrrrr. I'll have to revisit next week

ShikOfTheRa, 2018-07-30:
This is now availablefor testing:

MWOSD Release 1.9 RELEASE CANDIDATE for BETA testers
1.9 PC client GUI is required: https://github.com/ShikOfTheRa/scarab-osd-gui/tree/master/MW_OSD_GUI
New fonts will need to be installed from R1.9 GUI app
Default the settings before configuring

closed 2018-08-08 by ShikOfTheRa