ShikOfTheRa/scarab-osd #504: Airspeed not working with INAV 1.9

Dronek, 2018-04-05:
Please refer to [https://github.com/iNavFlight/inav/issues/3010](https://github.com/iNavFlight/inav/issues/3010)

I activated the airspeed indicator in MWOSD 1.8 and the airspeed sensor is connected to and logged correctly by INAV. However, the indicator in MWOSD constantly shows 0.0 km/h. Is this problem related to inav not sending the airspeed data over MSP or is this a MWOSD issue? Thanks in advance.

ShikOfTheRa, 2018-04-06:
It wasn't available at time of dev. Oly for Ardupilot/ PX4 or directly connected to certain hardware boards

Will be added as soon as I know when it's exposed via MSP!

ShikOfTheRa, 2018-04-06:
OK - re-read and see its in dev. Will add next week

ShikOfTheRa, 2018-04-06:
meh. Just looked at this and it's not available via original MSP.

Will add to the wish list. Not quite sure when - or if can implement.

ShikOfTheRa, 2018-04-06:
Added to roadmap: https://github.com/ShikOfTheRa/scarab-osd/wiki/Developer-guide

Not yet planned:
* iNav Airspeed (if feasible with MSP v2)

ShikOfTheRa, 2018-04-09:
Had a bit more of a look at this today. 
I see response from iNav is it must use protocol v2
Its a pain - none of the others do but if that's the only way we will have to look at that.

I will try to integrate, but bear in mind it is small 8 bit chip in OSD. I do not know at this point if there is enough memory to add handling of 2 protocols.
I will try - I would like to see it available for future. At least until iNAv OSD catches up and to help with support for other hardware that does not have embedded MAX.

Dronek, 2018-04-10:
Thanks for your effort.
I also noticed some non-functional indicators in mwosd 1.8 together with inav. Maybe it is also related to the protocol. Will check the configuration and do a test flight again to be sure before issuing here.

ShikOfTheRa, 2018-04-11:
yeah there are a few features such as windspeed, airspeed, wind direction, etc. that were intended for Ardupilot / PX4 users. If anything not working or is now supported by iNav, let me know.

Started working on this. Will be a few days...

Dronek, 2018-04-14:
I'm going to post all issues, I have with mwosd 1.8 and inav 1.9 in this issue here to avoid opening multiple issues. This is "mwosd 1.8 inav aero" version flashed by the recent mwosd configurator on minimosd. Flown on a flying wing.

- Somehow the baro alt indicator and the vario stopped working after switching from mwosd 1.7 to 1.8.  Only the GPS altitude works. In the attached image you can see the GPS alt indicator showing a valid altitude while the baro altitude is zero. The vario shows a constant zero as well. I checked the sensors in inav configurator and the baro works fine there.
As a side note, I chose "use baro alt and mag heading" in the mwosd configuration ... in the GPS tab if I remember correctly.

- The glidescope constantly shows the lowest position. It actually worked pretty well in mwosd 1.7. I suspect, this has to do with the baro altitude not working.

- This is a rather cosmetic issue: I have the FC mounted backwards and configured inav to account for this by setting "align_board_yaw = 180". The pitch indicator in mwosd shows negative values for pitch up and positive for pitch down so I guess it does not care about "align_board_yaw" (CLI command) in inav.

[image]

Dronek, 2018-04-15:
I also noticed that when switching osd layouts via a channel, the order of MID and HIGH seems to be mixed. I assigned different layouts in the configurator and when i switch the channel from low to high, the layouts go LOW - > HIGH - > MID.

ShikOfTheRa, 2018-04-16:
Thanks for reporting these. I'll get on to them tomorrow.

ShikOfTheRa, 2018-04-18:
Quick update - I have been tied up with work and will be away for next 8 days:

R1.8 bugs:

Confirmed/fixed - Ardupilot. Wind direction out by 180 - same for other minim firmware available. 

Confirmed - iNav. Baro alt displays 0 when baro used. GPS OK. 

Confirmed - iNav. Glidescope always shows lowest. 

Cosmetic - iNav. Pitch values incorrect when yaw alignment changed.

- I believe OSD and iNAV is working as intended. See comments below.
- Note this can be enabled in config.h using //#define INVERT_PITCH_SIGN
- Not sure there is any standard for pitch up being positive. 
- I will make pitch up being positive as it seems more popular. From next service pack 

Confirmed/fixed - iNav. OSD LOW/MID/HIGH are switched around.

kn8alpha, 2018-05-01:
Could you check on GPS DOP as well, please. Always shows 0.0 for me.

ShikOfTheRa, 2018-05-02:
Currently only supported for GPS OSD

I will add next week for iNAV 

Back from holidays so hope to clear the issues raised above

kn8alpha, 2018-05-21:
I can confirm that GPS DOP works with iNav 1.9.1 now

ShikOfTheRa, 2018-05-21:
Excellent - thanks for feedback.

Dronek, 2018-05-28:
Confirming after test flight with iNav:
- Baro Alt works; vario as well
- Glidescope works (BTW, what's the default angle setting?)
- Pitch-up now positive instead of negative
- OSD switch positions still seem to be mixed up; will double check

Thanks for fixing :)

ShikOfTheRa, 2018-05-29:
Excellent! thanks also for your testing time.

Default glidescope was 4 degrees, in latest version it is now 8 degrees after feedback from others. That seems a little to high, but i guess for most we are not talking about high performance gliders! But it is quite reliant upon home altitude setting being correct.

I hope to make it an adjustable parameter in future.

Interested in your thoughts on best default angle

ShikOfTheRa, 2018-06-20:
MSPV2 and airspeed fromiNAV implemented. I have no means to test, 
Unfortunately as now have to support MSPV2, it takes quite a bit of memory overhead. Something else will Have to be removed until I can save memory somewhere.
Suggest disable mapmode in config.h to free memory if not used

Avialble in config.h
//#define MSPV2                     // MSPV2 and Airspeed support

ShikOfTheRa, 2018-06-28:
@Dronek  if able to test I can provide new GUI to support latest dev release for airspeed testing.

Dronek, 2018-06-29:
I'll try to test it this weekend. Loosing map mode for airspeed is no option for me, however. I'm using map mode in almost every flight. Will play around with the config.h in order to find a good compromise.

ShikOfTheRa, 2018-07-30:
iNAV airspeed is now available for testing -  retaining map mode. I do not have kit yet to test:

MWOSD Release 1.9 RELEASE CANDIDATE for BETA testers
1.9 PC client GUI is required: https://github.com/ShikOfTheRa/scarab-osd-gui/tree/master/MW_OSD_GUI
New fonts will need to be installed from R1.9 GUI app
Default the settings before configuring

Glidescope - 
Some additional options in config.h I hope in future to free memory to migrate into the GUI.

ShikOfTheRa, 2018-08-08:
I have tested as far as can seems OK. Closing
reply to thread if required. I still get notifications.

ShikOfTheRa, 2018-08-08:
1.9 will be out this week.. It is nearly identical to beta release

closed 2018-08-08 by ShikOfTheRa

ShikOfTheRa, 2018-08-21:
Thanks for the feedback and your time testing. I'll have to get some hardware and test it. Its the only way! Have users with ardupilot running it so I hope it's only a small change needed. Set ot low priority :)

I like the Matek boards - good choice. I also tried the FFPV F35 as it looked good, but I'd take the Matek boards any day for a wing.

With 1.9 iNav OSD just not good enough for me so continued to run separate hardware / mwosd. With 2.0 its caught up and with the extra processing power / access to variables it's moved ahead. The guys have done a great job. 

I think the future of MWOSD is ardupilot and GPSOSD - 6 months ago they hardly showed in the stats, but now that’s changing significantly.
1200 installs this week - 281 betaflight, 268 iNav and 253 ardupilot.

Much that I like all the tech, have really enjoyed flying gpsosd only lately. A lot less hassle and a bit of a buzz from the extra risk of no rth.