ShikOfTheRa/scarab-osd #505: Incorrect Miles Long Distance Calculation

trailx, 2018-04-08:
`+#ifdef LONG_RANGE_DISPLAY  
+  if (d2f>9999){
+    if(Settings[S_UNITSYSTEM]){
+      tmp = ((d2f) + (d2f%5280))/528;
+    }
+    else{
+      tmp = d2f/100;
+    }
+    itoa(tmp, screenBuffer+units, 10);
+    uint8_t xx = FindNull();
+    screenBuffer[xx]=screenBuffer[xx-1];
+    screenBuffer[xx-1] = DECIMAL;
+    xx++;
+    screenBuffer[xx] = 0;
+    type = (type==2) ? type=6 : type=4 ; //          
+  }`

Miles is being calculated incorrectly in the screen.ino file, at line 2512.

`+      tmp = ((d2f) + (d2f%5280))/528;`

Should be:

` tmp = (((d2f)/528)+.5);`

This video explains it with more detail:

https://www.youtube.com/watch?v=_SIzaa_hUdc

Thanks!

ShikOfTheRa, 2018-04-09:
Great job - thanks. Will get on to this today

ShikOfTheRa, 2018-04-09:
OK - confirmed bug. 
Have submitted an update to main repository for testing.

Added rounding improvements as per your 0.5 suggestion. It never bothered me and always wondered when someone would pick it up!

trailx, 2018-04-09:
Thanks for taking a look!  Now I need to figure out how to flash my OSD.

ShikOfTheRa, 2018-04-09:
I can produce a hex for you if it’s easier? Inav fixedwing?

trailx, 2018-04-09:
That would be awesome!  I'll be out flying wednesday and can test then if you can make that happen!  I have this inav fixed wing OSD: http://www.carbonbird.com/products/inav-osd-for-air3-airhero32  Not sure if that makes a difference or not.

So, all I have to do is just upload using the flash button on the chrome MWOSD app, and load the hex file from disk?  I assume the osd settings will be unchanged before/after - or do I need to export/import?  (Thanks! - I try not to update firmware when things are working, so I'm always a little hesitant when I do)

ShikOfTheRa, 2018-04-09:
OK. Easier solution:
No change to settings...
Tick "Show test releases" in GUI
Should see 1.8.0.1

ShikOfTheRa, 2018-04-12:
confirmed resolved in main resository. Full release available shortly.

closed 2018-04-12 by ShikOfTheRa