ShikOfTheRa/scarab-osd #507: Can't write config to OSD from GUI. MSP simulation is working.

mazsola76, 2018-04-18:
MWOSD 1.8.0.0
FW. "Minim", loaded online
"Write" button is inactive despite many setting changes.
MSP simulation is working, on the other hand.
Am I missing something?

BR,
Zsolt

ShikOfTheRa, 2018-04-18:
Hi - have you registered? button under connect?

mazsola76, 2018-04-19:
OK, solved!
Maybe it would be a good idea to make a popup window for unregistered users to explain THIS software is not registered and what it means (not to pull out their hairs what's happening :)
Great software, the simulator is an awesome idea, but the buttons must be ironed out a bit  - for me they are shifting all over the main window, and are responsive mainly at their lower edges :(  - if at all.
Thanks and best regards,
Zsolt

mazsola76, 2018-04-19:
One issue remains:
Loading a previous config from a .json file is possible on the virtual OSD, but the "write" is failing.
Modifying as many parameters as you wish on the virtual OSD and uploading them works fine!

closed 2018-04-25 by mazsola76