ShikOfTheRa/scarab-osd #512: Wrong current and voltage reading with aeromax osd

shro70, 2018-05-27:
Hi.

I have a problem with my aeromax osd + pitlab current meter.
The 1.8 configurator show me the correct value but the osd show a voltage of 0.8 to 1.25v and a current of 0. I tried to play with the option in the GUI without success.

Config : 
-aeromax osd 1.8 inav aero firmware with putlab current meter 50a .

ShikOfTheRa, 2018-05-29:
Hi - please note the GUI preview is only for organising the screen - it doesn't replicate true value of OSD.

For voltage - that sensor has a divider on board. There is also one on aeromax so cant use that output. Connect 3s/4s/5s full battery voltage into the aeromax vbat pin. It is why you are seeing a low voltage - because it is already divided to a small value.

For current:
This should work.
Connect the "I" output from sensor
Make sure "Use direct connect" is enabled.
Offset = 0 to start
Adjust = 750 to start

shro70, 2018-05-29:
Hi. It's not the GUI preview which show me the correct value but the values in the upper part of the configurator.
I have enabled "direct connection" and "5v ref" and the "adjust" setting is limited to ~256.

Thanks for you reply and sorry for my bad English.

ShikOfTheRa, 2018-05-29:
Ah, I understand. What version are you running? I just tried the latest 1.8 and it allows me to set higher values

shro70, 2018-05-29:
I'm using the 1.8 version. I even tested the 1.8 1 test release without success.

ShikOfTheRa, 2018-05-29:
Hi - I can change OK - see example. Not you might have to connect to see true max values allowed.

[image]

ShikOfTheRa, 2018-06-13:
Did you find a resolution?

closed 2018-06-20 by ShikOfTheRa