ShikOfTheRa/scarab-osd #515: Throttle setting and Pitch value indicator

JeyPi, 2018-05-31:
I am using the MW configurator 1.8.0.0
PX4 + arduplane
Micro OSD v2.3

The pitch indicator uses negative numbers during climb and positive during descend. that is obviously nonsense.
The THR indicator reacts to my channel 3 while on my radio it is channel 1.
No matter what I set in the transmitter type.
Mine is mode 1
Channel 1 right stick --Up and Down THR
Channel 2 right stick -- L and R  Ailerons
Channel 3  left stick-- Up and Down Elevator
Channel 4 left stick --L and R  rudder

ShikOfTheRa, 2018-06-01:
Pitch value:
There is no right or wrong way. Some are positive forward, other negative.Same as roll or yaw.
I would agree that pitch up  = positive is more prevelant - and code default was recently amended to reflect that. It was a setting for those that compile their own and do not flash from the gui

TX type support:
This wasn't available for ardupilot/px4 users. It is now added for testing.

You can try release 1.8.1.2 which is available if select the "show test releases" in the gui flash utility

ShikOfTheRa, 2018-06-01:
If tick "test box" like below, should see test releases. it has changes which hopefully you need. I only uploaded this morning. Just pick latest the one.
 
[image]

To get correct throttle display, have to match following to how your tx is set up. Probably the middle one. This functionality wasn't enabled for px4/ardupilot in 1.8 of the OSD
[image]

ShikOfTheRa, 2018-06-01:
for pitch roll I could set defaults so that up/down are different for copter/plane - but I can't win as always someone wants it different :) probably not enough memory for it to be a gui option. 
I think up = positive for airplane and copter is best.

If you build own using arduino, these parameters can be changed as below:
https://github.com/ShikOfTheRa/scarab-osd/wiki/Custom-installation-guide

#define INVERT_PITCH_SIGN           // Invert the sign of the displayed numeric value for the pitch angle (ex: pitch up = positive )
//#define INVERT_ROLL_SIGN          // Invert the sign of the displayed numeric value for the roll angle (ex: roll right = negative )

JeyPi, 2018-06-03:
I was able to flash as you suggested and the pitch sign works fine for me as it is supposed to. I did not have to use Arduino to correct it. Also the THR problem solved Thanks. Closing

closed 2018-06-03 by JeyPi