ShikOfTheRa/scarab-osd #517: Baro Altitude AMSL or AGL (above mean sea level or above ground level)

JeyPi, 2018-06-04:
Newest version MWOSD
arduplane - pixhawk
Using the configurator in the GPS field I clicked the Baro alt + gps heading

Sometimes the altitude digits show altitude above the sea level and sometimes they show the height above the ground. (One or another during the whole flight). I was not able to figure out what is this determined by. Some might prefer that  and some this. It would be great to have that tick-able.  
I might be doing something wrong though. If so please advice thanks.

ShikOfTheRa, 2018-06-05:
Hi - would you be able to share a video and ideally a tlog with me to help diagnose?
I think from what you are saying, it reports consistently through flight, but is either one or the other... It doesn't swap between AGL/ASL during flight?
If that is the case, that gives me a bit of a clue - probably the home setting is incorrect.

All altitudes displayed should be relative to home launch. Effectively GPS used to calibrate QNH and the reported data is height above home launch position.  

I do get asked occasionally for options to display AMSL and it is on the todo list. For most 99% of FPV pilots and I think AMSL is not what they need :) myself included.... 

For mavlink based systems:
Baro aIt and gps alt should effectively read the same as it is taken from GPS data.  
If you have lidar / rangefinder installed, this can change how it works - although really should just show more accurate info up to range of the unit.

JeyPi, 2018-06-05:
That is right. It moustly shows the AMSL altitude. And sometimes AGL.
One or another througout the whole flight. No swapping.
I will make a video and ahare a tlog.
How do I influece the home point registering. Before I fly I will also update to the newest arduplane soft.

ShikOfTheRa, 2018-06-05:
This is MWOSD issue. 
Until I identify and provide a fix, if you check altitude altitude when arming, - if not showing approx 0m suggest to disarm/re-arm.

I have an idea of what might be the cause. I want to try to emulate so I can verify fixed.

JeyPi, 2018-06-05:
I tried ground tests. This time no zero out of 10 tries. No matter arming or disarming.
The MWOSD started with zeroed altitude but after the baro sensor got calibrated it jumped exactly to MSL. I was not able to replicate the situation when I had zero on the altimeter taking off.
I have the newest arduplane software in use. Also I noticed the wind indicator and speed was way too off but that needs to be checked in the arduplane software as last but one version it was indicating nonsense.

ShikOfTheRa, 2018-06-05:
Ok, I will re-test tomorrow. If you could capture tlog or arm/disarm when it does t work it would help.
Thanks for help in testing. It is difficult for me as I never used ardupilot..
I have to get a plane with it soon!

JeyPi, 2018-06-06:
Hi. I tested today. Started in a basement.
Until the gps signal obtained I got perfect baro altitude AGL. As soon the GPS lock obtained it displayed AMSL.

JeyPi, 2018-06-06:
Send me your email to 
j.pistek@seznam.cz
Video and tlog ready

JeyPi, 2018-06-16:
I closed it by mistake. No change so far. As soon as the arduplane gets GPS lock, AMSL desplayies on the OSD.

ShikOfTheRa, 2018-07-24:
Well i haven't been able to replicate this further once the intermittent bug was resolved..

Just a thought...  Is the fixedwing alt/heading set to anything other than "Use GPS altitude/ GPS heading"?

closed 2018-08-08 by ShikOfTheRa