ShikOfTheRa/scarab-osd #519: MWOSD Configurator doesn't load specific fw

gyoezoe, 2018-06-07:
Hi,

In configurator 1.8.0.0
when I try to load 1.8.1.0 fw online for microminimosd - betaflight -multirotor I get the following alert:

Load HEX failed: NoSuchKeyThe specified key does not exist.32320252/5dd85892-637d-11e8-8317-36ace5ab8688E727E3D7F1554D61EplF5UXEZqqX5iB5UvTNrMyODYqFHgLngyEV53MklEvgsjQkEtkAPOUNgC6yrcMotHf+iLVf/Us=

Other versions loaded fine.

thnx
gyoezoe

ShikOfTheRa, 2018-06-07:
wow that's a new one on me.
It looks like a problem with github hosting. Unable to download it directly either. I'll have to raise a support ticket.

gyoezoe, 2018-06-08:
It's working now, thank You

closed 2018-06-08 by gyoezoe

ShikOfTheRa, 2018-06-08:
Yes - it was github issue. I had response from them that there was an error. I removed and re-released and it seemed to work.

Thanks for raising