ShikOfTheRa/scarab-osd #521: Voltage issues

sevrugin, 2018-06-08:
Hardware:
APM 1.8
microminimosd connected to the FC telemetry

FC is powered by 3s or 4s battery
OSD is powered by custom 5v adapter

Issues: 
- Main voltage is show 3.2v with alarm 
  souce: FC voltage / autodetect cell count and cell alarm / autodetect cell alarm 3.4
- Video voltage is show 0v (I tried to switch between BAT1 and BAT2 pin)

If I choose voltage to the "DC source" then "Video voltage" is show correct data but main voltage is 0v

So question is - can I choose FC source for the main voltage and DC source for video voltage?

PS: On the 1.7 firmware it worked correctly

ShikOfTheRa, 2018-06-08:
Hi - I have amended - can test this by using firmware flasher. Select "show test releases" and pick 1.8.1.5 or later.

I am not sure about the first point - I see correct operation there. 
Might be worth setting defaults from gui after upgrade and re-connecting

sevrugin, 2018-06-08:
Great! I'll test in the evening

sevrugin, 2018-06-14:
Great work! 
Thanks for your fix! Now everything work as expected! 

Another question. 
Is there a way to add "m/s" to the Units of Measure? 
I'd like to see speed (AS and GS) and variometr on the m/s

ShikOfTheRa, 2018-06-15:
Yes - this is possible. I am happy to add this to the development list. It is minor so can do soon.

But... - it will be a custom install using arduino to edit/upload - not available in GUI as it is rarely used and unfortunately I do not have the memory to be able to support all features. I wish I could!

If you are happy wit this approach I do it for you.

ShikOfTheRa, 2018-06-15:
Oh - and thanks for testing and confirmation feedback!

sevrugin, 2018-06-15:
It would be great! Thanks!
No problem to use Arduino

ShikOfTheRa, 2018-06-20:
Added the m/s request into development road map. I plan to do this soon.

Think the issue can be closed

closed 2018-06-20 by ShikOfTheRa