ShikOfTheRa/scarab-osd #524: Changing pin mappings for airspeed (minimosd micro)

Tdogb, 2018-06-10:
I am not sure if this should go here, but I was wondering how I could change the pin mappings. I want to use my osd (minimosd micro) with an airspeed sensor (MPXV7002DP) which outputs an analog voltage from 0v-5v depending on the airspeed. I was looking into changing the pin mappings in Def.h by switching the AUXPIN to and BAT2 to . However, I haven't had any success. Do you guys have an idea of what I am doing wrong. Beginner here and this is only for my personal use, not for merging into the actual project.

ShikOfTheRa, 2018-06-11:
OK - I will have a look. I am not sure if airspeed is working for direct connect at moment. It was something I wanted to add.  Give me a day of so and I will check and verify...

ShikOfTheRa, 2018-06-13:
OK have looked at this as far as I can ...
Firstly, it was not completed / working. It is now working on the bench for me as far as I can simulate it using a voltage on the pins.

Notes:
- will require GUI to calibrate - for now its set to approx settings from datasheet. 
- I get a max speed of about 220kmh which seems about right
- voltage reference must be set to 5v
- will now default to using vbat2 pin unless using aeromax hardware which has a pin to support this.
- Airspeed is always set to zero when disarmed- I might change this. Find and remove  if(!armed) t_value=0; in :

void display_speed(int16_t t_value, uint8_t t_position, uint8_t t_leadicon)
{
  if(!armed) t_value=0;
 
Will get a GUI version to support out in next couple of weeks.

ShikOfTheRa, 2018-06-14:
My bad - default input changed to amperage pin - A1

Can't use vidvoltage/vbat2 pin as it has a voltage divider attached..... I forgot this as my test rig doesn't. It is hard wired.

ShikOfTheRa, 2018-06-20:
This will be available in next release of GUI . If need it prior to release, let me know...

Closing as completed

closed 2018-06-20 by ShikOfTheRa

jjameshodgson, 2018-07-09:
Has this been added to Test release 1.8.1.6 ? I am looking to connect an MPXV7002DP directly to the amperage pin of my micro minimOSD but don't see any options in the configurator to enable an airspeed sensor on this pin. Any help is appreciated, thank you and keep up the amazing work!

ShikOfTheRa, 2018-07-16:
Hello - sorry for delay, I have been on holiday.

Temperature probe / Airspeed sensor has been re-worked. It is also now better supported with configuration/calibration options in the GUI.

It will be out very shortly.... I hope by end of this week. It has been tested only with temperature sensor so far. 

Documentation for it is here ready
https://github.com/ShikOfTheRa/scarab-osd/wiki/Temperature-probe

https://github.com/ShikOfTheRa/scarab-osd/wiki/Airspeed-Sensor

I'll p[ost when its available to test.

jjameshodgson, 2018-07-16:
Ok thanks, since I didn't hear back I started diving into the source code and got it working on the AMPERAGE pin. But with no easy way of changing the EEPROM calibration and offset values with the old GUI I was forced to implement those parameters in config.h and it isn't particularly pretty nor easy to adjust afterwards. Thanks for adding this feature