ShikOfTheRa/scarab-osd #526: Need longitude and latitude display with full precision

han8jun, 2018-06-11:
The GPS position displaying format likes "30.23884N" and "119.97914E", mostly they are good enough. But sometimes we need more precision to locate the crashed aircraft ( use some map-tool based accurate longitude and latitude). An option of display format would be nice feature.

ShikOfTheRa, 2018-06-12:
Default display is 5 decimal places - accurate to approx 1 meter. I think this is enough for recovery?

https://en.wikipedia.org/wiki/Decimal_degrees

Can be tweked of course. I could add an option to define accuracy  in config.h for those who use arduino to compile/upload.

macfly1202, 2018-06-14:
Hi @han8jun, Have you got 20-40 meter deviation in reality or on a map ? I say that because from what I know, publishing map of China use it own kind of GPS coordinate that did not fit the google one for example. Don't know if linked...

ShikOfTheRa, 2018-06-14:
This is interesting to find out.

I think in China it will be difficult to get accuracy unless you use correct method. I just read link below. 
Increasing numbers displayed will not help - it does not change location by more than 1m.

https://en.wikipedia.org/wiki/Restrictions_on_geographic_data_in_China

han8jun, 2018-06-19:
@macfly1202 , It was in a map. I realized the problem was maybe about the maps, then I made an android APP which ignores the POI draw on background map, It's just focus 2 relative coordinates between the airplane and the mobiphone. We finally found ".DDDDD" precision was OK, the daviation were within 1~3 meters.

han8jun, 2018-06-19:
Based a new test mentioned in last reply, seems dont need to make any change of osd codes. @ShikOfTheRa .

closed 2018-06-20 by han8jun

ShikOfTheRa, 2018-06-20:
Thanks for the update and feedback.

It was interesting to learn...