ShikOfTheRa/scarab-osd #528: Long lines breaking arduino IDE

ledvinap, 2018-06-13:
Lines  in font?.h are too long, probably breaking Arduino IDE function signature generation. 
Build then fails with 'checkEEPROM' was not declared in this scope.

Splitting the line roughly in half will fix the problem.

ShikOfTheRa, 2018-06-13:
Ah yes - had this before. Correct - I had to split the lines. I updated fonts and forgot....
If works in platformio :)

i'll update....

In GUI flasher - utilities, fonts there are precompiled ones

ShikOfTheRa, 2018-06-13:
Also to add - I think it depends upon Arduino / compiler version
1.6.5 seems to work OK. Vagualy recall this too.

ledvinap, 2018-06-13:
I tried 1.6.something and 1.8.5, both fail
It's just one of many annoying Arduino IDE bugs/features ...

ShikOfTheRa, 2018-06-15:
I've updated in main repository. Thanks again for pointing out.

It worked for me with single line in 1.6.5 and also platformio - but I know I definitely had same issue.

ShikOfTheRa, 2018-06-20:
Ok think this cam be closed- please re-open if not resolved.

I'll try to remember to split files in future...

closed 2018-06-20 by ShikOfTheRa