ShikOfTheRa/scarab-osd #529: Invalid handling of temperature

ledvinap, 2018-06-13:
https://github.com/ShikOfTheRa/scarab-osd/blob/master/MW_OSD/MW_OSD.ino#L1245-L1246

Due to operator priority, expression is evaluated as `sensorfilter[3][SENSORFILTERSIZE]>>(3-TEMPZERO)`
(gcc warns about this. Nobody is reading warnings with Arduino?)

when fixed, TEMPZERO gets subtracted twice (`map` has `TEMPZERO` as lower input bound)

Both problems are invisible when `TEMPZERO` is `0`

ShikOfTheRa, 2018-06-13:
Yes, correct - never used it myself as directly connected. You may be the first. 
I'll get it amended.

There are a lot of warnings that need to be cleared up. On the to do list when get time...

ShikOfTheRa, 2018-06-15:
Updated in main repository. 
Thanks for pointing out!

ShikOfTheRa, 2018-06-15:
Just to add I'll run through the compiler issues in a couple of weeks. See what can clean up

ShikOfTheRa, 2018-06-20:
Ok think this cam be closed- please re-open if not resolved.

closed 2018-06-20 by ShikOfTheRa