ShikOfTheRa/scarab-osd #532: Throttle indicator: add option to show actual output

tyllurius, 2018-06-19:
Hi!
The throttle indicator seems to show the operator value (RC channel input), which makes sense for racequads. But in auto-flight modes this value is mixed/overridden by the autopilot. So for fixed wings which are often used in auto modes it would be nice if there were an option for the throttle indicator showing the actual output to the ESC (RC channel output).

ShikOfTheRa, 2018-06-20:
The option below may work in config.h

I'll try some simulation. I am not too familiar with ardupilot/px4

//#define MAV_ALT_THROTTLE          // Use alternative MAV throttle value. Not raw RC channel.

ShikOfTheRa, 2018-06-20:
I just tried theis. MAV_ALT_THROTTLE  was not implemented.

I have resolved and with some test tlog data, it seems to work as you request. In cruise mode I see throttle value change as it maintains altitude, but RC value stays consistent.

In next release this will be default.

Also note on GUI you can chosoe to display throttle % or PWM value...

ShikOfTheRa, 2018-06-20:
This is now available in main repository for manual installation.
Will be available in bundled releas ein couple of weeks.

Think this can be closed

closed 2018-06-20 by ShikOfTheRa

tyllurius, 2018-06-20:
Excellent, thanks! Will try out as soon as possible and looking forward for next release!