ShikOfTheRa/scarab-osd #533: Efficiency indicator: add mAh/km

tyllurius, 2018-06-19:
Hi!
This was already discussed (#430 , #373 ), existing options are good to check efficiency, but for mental maths during long range flights (rough estimation how many remaining mA/h you need to get back) it would be useful if it showed mAh/km, so it's easier to refer it to the mAh counter indicator.

ShikOfTheRa, 2018-06-20:
I get so many variations of efficiency requested!

I think it is this one...
mA/h required to travel xx km to get back to home. Work out using known values - xx km to home and average mA/km during flight to date? 

Requires accurate current sensor calibration. This method is independent of battery capacity is positive)

tyllurius, 2018-06-20:
Yes, a lot of discussions around that ;)
For some people mAh/km is more useful to calc the flight time left, because you know how far you are but not how fast you can/want to go for the flight back.
Accurate current measuring is available.

ShikOfTheRa, 2018-08-08:
This is added in main repository. 
Will need the beta test GUI to configure.

If can wait a few days, R1.9 will be out very soon!

closed 2018-08-08 by ShikOfTheRa