ShikOfTheRa/scarab-osd #534: Add configurable sensor indicator icon

tyllurius, 2018-06-19:
Hi!
I'm looking for a way to display the values of sensors attached to the input pins (V1, V2, Amp, RSSI). The sensors can be a different (analog or PWM value), but I think it can be implemented with one solution.
I will try to explain it by using my two purposes; the current status is: Voltage, Amperage and RSSI from FC, so free OSD pins can be used to obtain additional sensor data. It would be nice if it could:
1. show value of an analog sensor, e.g. external temperatur sensor ( #457 ). The pins value should be [adjustable](https://www.rcgroups.com/forums/showpost.php?p=39286885&postcount=2778) and be mapped to a [GUI indicator icon](https://www.rcgroups.com/forums/showpost.php?p=39266174&postcount=2776)
2. also read PWM values, for example a RSSI value in PWM format, to be capable of displaying multiple RSSI values. Some modern receivers can output two different RSSI values, real RSSI and LQ (link qality). So the OSD can obtained the first value from FC data stream and the second value can be fed as PWM into its pins (no more need for analog conversion).

ShikOfTheRa, 2018-06-20:
Notes:
Inputs for V1 and V2 nor normal OSD have a resistor divider - not able to use them for PWM etc.  Not good for sensor input because very low resolution.

There is support for directly connected temperature sensor - it is untested. In five years I think no-one has used.

There is support for reading PWM and PPM. Configuarble on the GUI. Actually I use both. I also use  directly connected LQ from EZUHF for my RSSI.

For 1 - This is a nice to have for sure. I would liek to take this approach.I have added to the development road-map wish list. In truth I do not know if/ when it will be completed. I do intend to revisit the reading of sensors as although it works perfectly it is a bit messy from growth development rather than strategic development.

For 2 - As there is a shortage of memory and this is rarely used I have added it as an option in config.h DUALRSSI in development tree at moment. Will be in next release shortly. Need some real world testing. Works in lab.

tyllurius, 2018-06-20:
Yes, I've read about the voltage divider..so, if I understood correct, for useful analog measurement only RSSI and Amperage pins make sense? 
Ok if the code for temp sensor support is existing I'd like to try it out.

1./2. Pleased to read that this functions are already existing in basic, so I'm looking forward for next release as I can help with real world testing!

tyllurius, 2018-07-08:
Hardware Temp Sensor: I can confirm this working. Hooked up a TMP36 on Amp Pin, to show up the value I  had to comment out //#elif defined PROTOCOL_MAVLINK in temperature section of MW_OSD.ino . But the icon keeps blinking (Alarm?).

DUALRSSI: Working excellent! Tested on flight with PWM RSSI on RSSI pin and an second from FC (v1.8.1.7). Only bug is that the second line of the icon is always displaying even if switched off on screen layout.

ShikOfTheRa, 2018-07-24:
Here is latest dev which has temperature probe. Also DUALRSSI bug fixed (thanks for advising. Sorry for delay - been on holiday):
https://github.com/ShikOfTheRa/scarab-osd/archive/Dev.zip

Here is latest GUI which has support for it. Adjustable on the GUI:
https://github.com/ShikOfTheRa/scarab-osd-gui/tree/master/MW_OSD_GUI/R1.9%20beta

Instructions:
https://github.com/ShikOfTheRa/scarab-osd/wiki/Temperature-probe

This will be close to the 1.9 release candidate

Example video of current sensor - results when attached to a VTX. 
https://www.youtube.com/watch?v=2PP7rqV95Qw

ShikOfTheRa, 2018-07-30:
This is now available for testing as a beta test release. DualRSSI is likely to remain a custom option due to resources I have available:

MWOSD Release 1.9 RELEASE CANDIDATE for BETA testers
1.9 PC client GUI is required: https://github.com/ShikOfTheRa/scarab-osd-gui/tree/master/MW_OSD_GUI
New fonts will need to be installed from R1.9 GUI app
Default the settings before configuring

ShikOfTheRa, 2018-08-08:
R1.9 will be out this week incorporating changes.

closed 2018-08-08 by ShikOfTheRa