ShikOfTheRa/scarab-osd #538: 1.8 on arch linux

mfulz, 2018-07-20:
Getting an segfault on arch linux :(

traps: MWOSD configura[14332] general protection ip:7f6ae84ddd60 sp:7ffd10912438 error:0 in libpangoft2-1.0.so.0.4200.1[7f6ae84d1000+13000]

mfulz, 2018-07-25:
The chrome APP Version is running 1.7 i think.
There's no other GUI Version previous 1.8 for Linux.

And as there are no sources available it's hard to help Findung the root cause oft the issue

ShikOfTheRa, 2018-07-25:
Can try this one. Compatible with 1.8. If still no luck I will discuss with the developer..

https://github.com/ShikOfTheRa/scarab-osd-gui/tree/master/MW_OSD_GUI/R1.9%20beta

mfulz, 2018-07-26:
The same issue.
The win version with wine is not working either :(

error414, 2018-07-28:
Hmm :( it works for me 
[image]

I downloaded it from https://www.osboxes.org/arch-linux/#arch-linux-201708-vbox

mfulz, 2018-07-28:
Could you paste an ldd MOWSD configurator vor however the executable is called?

I would Mike to compare the library versions

error414, 2018-07-28:
https://pastebin.com/kQ321E1d  is it what you wanted?

mfulz, 2018-07-28:
Yes thanks will check that out

andrejpodzimek, 2018-08-02:
```
Thread 1 "MWOSD configura" received signal SIGSEGV, Segmentation fault.
0x00007fffe9e36d60 in pango_fc_font_get_glyph () from /usr/lib/libpangoft2-1.0.so.0
(gdb) bt
#0  0x00007fffe9e36d60 in pango_fc_font_get_glyph () at /usr/lib/libpangoft2-1.0.so.0
#1  0x00007fffe9e36db7 in  () at /usr/lib/libpangoft2-1.0.so.0
#2  0x00005555555813d4 in hb_font_funcs_set_glyph_extents_func ()
#3  0x0000000000000000 in  ()
```

This^^^ is with the file from the 1.9 beta (still called 1.8 for some reason). The 1.8 crash looks similar. I tried this with pango 1.42.1 and pango 1.42.3 and it segfaults in both cases.

(Calling a thread `MWOSD configura` already looks like some nasty memory corruption, but that's most likely a red herring. :grinning:)

andrejpodzimek, 2018-08-02:
https://pastebin.com/8y45hT22

error414, 2018-08-02:
Can you try this build please? https://www.dropbox.com/s/npj95fuy98v9a91/linux64.zip?dl=0

mfulz, 2018-08-02:
Great this one works.
Thanks a lot !

ShikOfTheRa, 2018-08-08:
Closed - thanks all! 
See if we can incorporate into R1.9 build shortly

closed 2018-08-08 by ShikOfTheRa