ShikOfTheRa/scarab-osd #543: Release r1.9 issue tracker

ShikOfTheRa, 2018-08-08:
Please report any issues identified in this tracker.

OSD:
None identified currently

GUI:
None identified currently

Enhancement requests (Transferred to development roadmap)

dwoolworth, 2018-10-31:
I'm running Arduplane v3.9.2 and using MWOSD 1.9 on MinimOSD unit.  Seems although I've selected "Imperial" as the units of measure, while MPH and Feet/Miles are working properly, the Altitude is still in meters, although it shows "ft" after the reading.

ShikOfTheRa, 2018-10-31:
@dwoolworth - bug confirmed. Thanks for reporting. My erorr - bug introduced after significant code changes to reduce memory for new features. Annoyed I missed that. . It should be fixed in 1.9.0.6 onwards - currently available if select "show test releases".

dwoolworth, 2018-10-31:
@ShikOfTheRa - It's all good!  Thank you for such an awesome product and pretty much the best OSD available!

ShikOfTheRa, 2018-12-07:
R1.9.1.1 is release with all reported bugs hopefully addressed. 

Closing this release tracker

closed 2018-12-07 by ShikOfTheRa