ShikOfTheRa/scarab-osd #544: Cannot flash BeeRotorF3 AIO (BRF3) (timeout waiting for bootloader after 400ms)

andrejpodzimek, 2018-08-08:
First and foremost, this is most likely not a MWOSD bug, because the Arduino GUI exhibits roughly the same problem. But having searched and asked around thoroughly, posting here is one of the last options I could think of. If there's a better place for this question than a GitHub issue (e.g. a forum), I'll be more than happy to post it there.

I can't flash the built-in MWOSD on [BeeRotorF3](http://rctimer.com/?product-1527.html), a historical / vintage :grin: ~2016 flight controller. I've described the struggle on rcgroups, [starting with this post](https://www.rcgroups.com/forums/showthread.php?2598047-BeeRotorF3-AIO-%28BRF3%29-Flight-Controller-%28integrated-OSD%29-Discussion-Thread/page86#post39972143).

(I'd like to update it to 1.8+, because it comes with 1.6 and is no longer compatible with the 1.8 GUI.)

In summary, the MWOSD GUI configuration works fine, but I can't flash it. Tried multiple FTDIs, 5V and 3.3V, even multiple different capacitors between DTR and RESET, double-checked DTR and RESET with an oscilloscope (to see if there's a pulse on the RESET side of capacitors) etc., but to no avail.

There are at least two [similar](http://intofpv.com/t-beerotor-f3-w-osd) [cases](https://nathan.vertile.com/blog/2016/02/03/rctimer-beerotorf3-flight-controller/) described on the web, but both unresolved. So I ended up with 3 specimen of the BRF3, all with the same problem. I do have a number of shiny F7 FCs that I could just swap in and forget about all of this, but I'd like to **understand** what's happening and why.

ShikOfTheRa, 2018-08-08:
Yeah I know. Good to understand. i still run a number of F3 boards. They work just as well as the F7 for me...
Also still need a bit more development on the built in OSD for me. It's getting there slowly. On the bigger planes I am still running external OSD even with internal ones available. 

Not seen, but this is what I would do based upon experience with similar types...

1 - I would disconnect everything else whilst flashing - just in case something else connected to the same tx/rx pins
2 - no other power than than provided by the ftdi. If board is powered, the reset required doesn't take place
3 - to me, the picture that shows the programming tx/rx wires seem the wrong way around from most ftdi programmers - so I would also try swapping rx/tx so they cross over

andrejpodzimek, 2018-08-08:
> 1 - I would disconnect everything else whilst flashing - just in case something else connected to the same tx/rx pins

:heavy_check_mark: Tried that. I bought 2 new BRF3s from rctimer and tried to flash those, because I thought my old one was faulty. But the problem was exactly the same, even with everything disconnected, just a bare new BRF3.

> 2 - no other power than than provided by the ftdi. If board is powered, the reset required doesn't take place

:heavy_check_mark: Sure, the MWOSD wouldn't work if the board had another source of power; it only works when both the STM32 MCU and the USB Silabs chip are completely off, because they all connect to the same UART1 lines. So I'm well aware of this requirement.

> 3 - to me, the picture that shows the programming tx/rx wires seem the wrong way around from most ftdi programmers - so I would also try swapping rx/tx so they cross over

:heavy_check_mark: MWOSD GUI (the old, Chrome-based one, compatible with older versions) works perfectly fine for me over the FTDI and connects to the BRF3 with no issues. (That's also the only way to connect to the MWOSD; one can't connect to it over USB on BRF3, because USB VCP is on UART1, the very same port that talks to MWOSD in flight, so USB + serialpassthrough is a no-go.) The GUI can read/write the MWOSD configuration and it successfully uploads fonts. So I'm pretty sure that TX and RX are connected (crossed) the way they should be.

There's one weird thing that I noticed on the FTDI, on its LEDs. It has two small LEDs monitoring TX and RX activity. When I try to flash MWOSD from the GUI, what I see is:

1. Two [DTR->RESET pulses](https://www.rcgroups.com/forums/showpost.php?p=40021447&postcount=1283).
2. A short TX activity around that time.
3. The 400ms timeout hits right afterwards.
3. A longer (1&ndash;2 seconds) RX activity occurs **after** the timeout.

So it might be the case that the MWOSD GUI (and also the Arduino IDE) waits for a response of some sort, but doesn't get it in time. Would it be feasible to extend that reset timeout, just to make sure it's not something trivial like not waiting long enough?

Yet another option that crossed my mind would be a faulty diode somewhere, which would lead VCC from the FTDI where it shouldn't go, e.g. into the USB Silabs chip. But I couldn't find 5V (or anything obviously non-zero) anywhere around the Silabs chip and/or around the STM32 MCU, so this hypothesis is likely also wrong.

ShikOfTheRa, 2018-08-09:
OK. Good troubleshooting there – and that can connect with old chrome GUI says basics are working right. 

I don’t know if this is an option, but on some, if you click flash in GUI or Arduino, then immediately plug in the OSD board it gets round weird DTR/reset issues. These are typically boards where manufacturer didn’t do the DTR / gnd correctly. There are a few 😊

What OS are you using? Windows?

error414, 2018-08-09:
Are you sure that it uses the arduino bootloader? 

Have you tried use this tool http://jbquad.blogspot.com/p/blog-page_10.html ? I think I can increase timeout to 1s but I am not sure that it helps.

ShikOfTheRa, 2018-08-09:
That is good point - if they haven't put bootloader on then will not be able to program via serial. Only using that tool or icsp pads

andrejpodzimek, 2018-08-09:
> I don’t know if this is an option, but on some, if you click flash in GUI or Arduino, then immediately plug in the OSD board it gets round weird DTR/reset issues. These are typically boards where manufacturer didn’t do the DTR / gnd correctly. There are a few :blush:

I did try the "click and plug" approach a few times, but not very thoroughly, because I didn't expect it to be a viable option. I'll give it a few more tries then. :smile: The FTDI connector is very easy to handle, so it should be doable even within the 400 ms, theoretically.

> What OS are you using? Windows?

Linux.

> Are you sure that it uses the arduino bootloader?

I have no clue. Because [the OSD manual for BRF3](http://www.helipal.com/download/uav/helipal-beerotor-brf3-osd-manual.pdf) explicitly mentions Arduino, I just assumed it would come with a compatible bootloader.

I must admit that I don't even know whether you can have MWOSD installed on the ATmega328p without having an Arduino bootloader there first.

> edit: this picture indicates it might have ISP option via 6 pin block. I usually make my own, but can purchase a ISP pogo adapter:
> http://rctimer.com/images/goods/20160805/c02c87817cce646e.jpg

Wow! And I've been wondering what those 6 pads were for. :grin: They weren't mentioned in any of the schematics. OK, so then I'm ordering [this](https://www.sparkfun.com/products/11591) and also [this](https://www.sparkfun.com/products/11801) &mdash; hope that's a compatible tool chain.

If I understand [this howto](https://learn.sparkfun.com/tutorials/tiny-avr-programmer-hookup-guide/programming-in-arduino) correctly, I could select the USBtinyISP programmer and burn a bootloader from the Arduino GUI. That could then get me into a situation where I cloud flash the BRF3 with just a FTDI, without special tools. Well, this is getting complicated, but at least I'll learn something.

ShikOfTheRa, 2018-08-09:
I'm just guessing on the pads!

For bootloader. Its not a quick thing to describe, but in brief arduino bootloader is not put in the default bootloader location and the atmega chip is set up to look at that new location. This enables arduino programmming over serial and preserves the arduino bootloader.
What can happen is a board can be programmed in  a different way which means it can only be programme using ICP. At least unless a new bootloader is put on.

mmm linux... 

any chance you could borrow and with a windows pc not a VM under linux? I'd definitely try that first

andrejpodzimek, 2018-08-09:
> I'm just guessing on the pads!

I'll double-check if they have continuity towards the right pins on the ATmega chip. A quick glance at the traces suggests that they do.

> What can happen is a board can be programmed in a different way which means it can only be programme using ICP. At least unless a new bootloader is put on.

Do I understand it correctly that if I set up the USB ISP programmer, then I can do the "burn bootloader" step from the Arduino GUI and get it into a bootloader configuration that lets me flash it over serial / FTDI?

Actually, if I could update MWOSD only through the USB ISP tool, it wouldn't be that much of a burden; I'd go that way if need be. But then I guess there would be no way to update it through the MWOSD GUI; there would always be some Arduino GUI magic&hellip;

> any chance you could borrow and with a windows pc not a VM under linux? I'd definitely try that first

I don't have access to such a machine at home or at work, so that would be quite hard to do. Also, in this case the [AVR programmer setup](https://learn.sparkfun.com/tutorials/tiny-avr-programmer-hookup-guide/?_ga=2.97958495.1538421370.1533821007-509136637.1527503040#driver-installation) looks way easier on Linux, becase (as usual) there's no "driver" installation step.

ShikOfTheRa, 2018-08-09:
>Do I understand it correctly that if I set up the USB ISP programmer, then I can do the "burn bootloader" step from the Arduino GUI and get it into a bootloader configuration that lets me flash it over serial / FTDI?

Yes! - as long as the manufacturer hasn't written any protection fuses.

>Actually, if I could update MWOSD only through the USB ISP tool, it wouldn't be that much of a burden; I'd go that way if need be. But then I guess there would be no way to update it through the MWOSD GUI; there would always be some Arduino GUI magic&hellip;

I've overwritten by accident - can still use GUI as normal through serial - just can't flash. Sort of what you are seeing !

Its a great way to learn ;)

andrejpodzimek, 2018-08-14:
So, just for the record and for future reference, the Sparkfun pogo module is utterly useless; it simply doesn't fit onto that pin header. First, the pin distance is different. Second, the pinout is completely different. So it looks like I'll have to solder small wires onto the pads and then figure out the right connection to the ISP module somehow, probably based on checking continuity towards the pads against the ATmega chip's pinout.

andrejpodzimek, 2018-08-14:
Allright, soldering done. Well, it doesn't work, because one probably can't burn the bootloader using usbtiny. So I guess I need a different AVR programmer. But which one then?

```
avrdude: Version 6.3, compiled on Aug  9 2017 at 12:31:56
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/etc/avrdude.conf"
         User configuration file is "/root/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : usbtiny
avrdude: usbdev_open(): Found USBtinyISP, bus:device: 004:064
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : USBtiny
         Description     : USBtiny simple USB programmer, http://www.ladyada.net/make/usbtinyisp/
avrdude: programmer operation not supported

avrdude: Using SCK period of 10 usec
avrdude: initialization failed, rc=-1
avrdude: AVR device initialized and ready to accept instructions
avrdude: Device signature = 0xa09332
avrdude: Expected signature for ATmega328P is 1E 95 0F

avrdude done.  Thank you.
```

andrejpodzimek, 2018-08-14:
Actually even the basic sanity check won't pass:

```
$ avrdude -F -c usbtiny -p atmega328p

avrdude: initialization failed, rc=-1
avrdude: AVR device initialized and ready to accept instructions
avrdude: Device signature = 0x20cbae
avrdude: Expected signature for ATmega328P is 1E 95 0F

avrdude done.  Thank you.
```

Could it be the case that the chip comes utterly broken on the BRF3, with some fuses set that the ISP programmer cannot overcome (e.g. RSTDISBL)?

Also, the "device signature" is *different* on each invocation and always different from the expected ATmega328p signature. But that's not surprising, because if the initialization failed, the `-F` is probably causing it to get some random garbage.

andrejpodzimek, 2018-08-14:
[image]

This^^^ is the setup. (The capacitors are a leftover from  past attempts to make the DTR <—> RESET pulse longer on serial.) Butcause the ATmega328p on the BeeRotorF3 appears to be broken or fused somehow, I’m sharing this mostly for the record. The pinout of the weird small header on BRF3 (orientation shown in the photo) is:

| | left | mid | right |
| :---: | :---: | :---: | :---: |
| **top** | MISO | MOSI | RST |
| **bottom** | 5V | SCK | GND |

andrejpodzimek, 2018-08-14:
# I flashed it! :joy: It’s so awesome!

Thanks a lot for all the advice. Here's a summary of the problems and solutions:

1. This.
    [image]
    Shockingly, I didn’t spot that mistake. I found it by taking an identical and unconnected programmer and checking continuity between all the 6 outputs and the chip socket. It didn’t match. Interrupting it was all that was needed.

2. It didn’t have a bootloader, indeed. Someone in the factory (or whoknowswhere) probably saved a few minutes and flashed it without the bootloader. :angry: That’s why it wasn’t flashable over the serial port.

That’s all. I got a bootloader there using the Arduino GUI and all Arduino stuff started working. First I flashed the `eeprom_clear` example. That worked. Then I went to the MWOSD GUI and flashed 1.8.x normally over serial! It flashed, all green, connected — and MWOSD seems to be working (minimosd).

Phew. This was The Worst Struggle. Ever.

What remains to be determined is whether the extra DTR <—> RESET capacitor is necessary (1 μF instead of 4 nF at the moment). Maybe not. I have one last BeeRotor F3 in a pristine state and would like to test it on that.

closed 2018-08-14 by andrejpodzimek

andrejpodzimek, 2018-08-14:
For the record, [I’ve updated the rcgroups thread](https://www.rcgroups.com/forums/showthread.php?2598047-BeeRotorF3-AIO-%28BRF3%29-Flight-Controller-%28integrated-OSD%29-Discussion-Thread/page87#post40078298) with more detailed instructions.

ShikOfTheRa, 2018-08-14:
Wow – you are very persevering and stubborn. You sound like me!! Honestly I am impressed – not many go to that length…
It’s a way to really learn more though!!

So no bootloader – that is bad. It is big oversight by manufacturer really. But it is easily done… I have done it myself. Even over serial it is possible to overwrite bootloader when flashing hex files. 

Interesting with Sparkfun pogo adapter - it is standard size. Maybe the Beerotor has non standard? Perhaps to fit on small board. I use Arduino uno and make my own pogo adapters with cheap pins from ebay.

Great job man. I can tell you are pleased!!

andrejpodzimek, 2018-08-15:
> What remains to be determined is whether the extra DTR <—> RESET capacitor is necessary (1 μF instead of 4 nF at the moment). Maybe not. I have one last BeeRotor F3 in a pristine state and would like to test it on that one.

The extra capacitor is **not** needed. The default 4nF are enough to reset the ATmega328p over DTR. Just tested it with a brand new BRF3. The bootloader was the only problem; there was no hardware issue.

andrejpodzimek, 2019-04-27:
For historical record (albeit somewhat off-topic), on-board MWOSD on `SPRACINGF3` [is also flashable](https://www.rcgroups.com/forums/showthread.php?2906663-Eachine-XFX90-90mm-Brushless-FPV-Racer-F3-OSD-Dshot-%28review%29#post41722327)! This is yet another vintage drone with on-board MWOSD where one can enjoy the l8test and gr8test versions. But this time it was easy, because the bootloader was there. It can be flashed using the GUI; no problem at all. (For the worst case, it does have pads for an ISP tool.)

The DTR <—> RESET capacitor value is 100 nF in this case, as opposed to 4 nF in the BRF3.

ShikOfTheRa, 2019-04-29:
Thanks for the continual update.. There are a lot of those boards out there.