ShikOfTheRa/scarab-osd #546: 1.9.0.0 Unable to WRITE select ICONS once moved from their default position in preview window

1280MHz, 2018-08-19:
I had to roll back to VERSION 1.8.1.0 to have the data displayed in my preferred screen format as I believe there is a bug in firmware version 1.9.0.0

POSSIBLE FIRMWARE BUG 1.9.0.0 
Firmware version: 1.9.0.0 (1900)
Device: AEROMAX
Variant: Ardupilot (APM)
Type: Aero
MWOSD Configurator version 1.9.0.0 

After successfully flashing the AEROMAX OSD with firmware v 1.9.0.0 via MWOSD Configurator v 1900, I was **unable to “WRITE” the “VOLTAGE – MAIN” & “DIRECTION TO HOME” icons after moving them from their default position within the configurator “preview window”**… (hangs at 29% “WRITE FAILED”). I believe there are more icons that cause the write to fail.

I loaded other default skins, saved it to the TX I was working in and then wrote everything to the board… restarted and the same issue occurred. Using MWOSD configurator 1.8.0.0 running firmware version 1.9.0.0 the error is still there. After uploading firmware version 1.9.0.0 via MWOSD configurator 1.8.0.0 the issue is still there.

BTW: Your AEROMAX OSD & MWOSD Configurator are fantastic… the benchmark.

Can you please inform me if there is a bug in firmware version 1.9.0.0?

All the best to you Shikra, looking forward your response.

janjazda, 2018-08-20:
I have exactly same issue. I'm using MWOSD R1.9 on MinimOSD and the newest GUI. 

I was unable to “WRITE” the **“DIRECTION TO HOME, THROTTLE, AP STATUS”**. 
"WRITE FAILED" occur and the icons getting back to default position. I didn't check all icons

ShikOfTheRa, 2018-08-20:
Hi - bug confirmed. Please stay tuned

1280MHz, 2018-08-20:
Hi ShikOfTheRa,

I really appreciate your prompt reply and confirmation of a 1.9.0.0 firmware bug.

Sh0tek61, 2018-08-21:
Same problem as described by janjazda also with "Voltage main" and "Sensors" - roll back to fw1.8.1.1 ok

ShikOfTheRa, 2018-08-21:
Hello - please try updated GUI here and let me know if it works. I have tested as far as posisble and it seem sOK:

https://github.com/ShikOfTheRa/scarab-osd-gui/tree/master/MW_OSD_GUI/R1.9

janjazda, 2018-08-21:
I can confirm that now everything is fine. New position of the icons can be write. 

I'm notice one more thing earlier. Glidescope is selected in GUI but isn't present on screen. Can someone have the same problem?

ShikOfTheRa, 2018-08-21:
I think it is only available to amend in fixedwing. Is that correct?

janjazda, 2018-08-21:
I think so. Currently I'm using MWOSD in fixed wing with Inav. 
Before update I was using R1.7.2 version and Glidescope was present.

ShikOfTheRa, 2018-08-22:
Hi - just tried this morning and it seems to work OK for me. 1.9.0.0 with starting point of default settings.

Note, it will only be visible if aircraft type=fixedwing as its not an available option on multirotor. Maybe in future will enable it for both. So if do not connect first, option will probably be hidden. Maybe that is it?

[image]

1280MHz, 2018-08-22:
Hi Shikra,

Thank you for correcting the issue and promptly releasing MWOSD Configurator firmware revision 1.9.0.1

Issue (as per my initial post) tested and resolved by R1.9.0.1 

Hi Janjazda, I also confirm glidescope is working correctly running Pixhawk 2.1 MWOSD R1.9.0.1 AEROMAX FW 1.9.0.0

All the best to you Shikra.

ShikOfTheRa, 2018-08-22:
Great. Thanks for the feedback. I'll let the GUI dev know...

@janjazda if you are still seeing issues can you raise a new issue. I'll close this one in couple of days and put out an new upgrade notice for the guys who are on the first 1.9 gui