ShikOfTheRa/scarab-osd #548: osd show text alarm "LOW SATS" when no GPS is configured in inav

unitware, 2018-09-08:
The only way to get rid of the flashing "LOW SATS" seems to be to hide text alarms, but then I seem to lose the DISARMED/ARMED text also.

MWOSD 1.9, INAV 1.7.3

ShikOfTheRa, 2018-09-10:
Sorry for delay - been away.

Yes - surprised this hasn't come up before. I'll look to add an option to disable automatically.

The sat message currently can't be disabled via the GUI however is an option if you use arduino custom install method and disable the following with a // in front of the line in config.h: 

//#define ALARM_SATS                  // Text alerts if sats below MINSATFIX - in addition to flashing sat indicator

ShikOfTheRa, 2018-09-10:
Another option - flash multirotor version.

Prime difference between multi and fixedwing version is to force the use of GPS sensors over baro.
I think that might work for you. Hopefully it will detect no gps installed and supress alarm

unitware, 2018-09-10:
Thanks, I'll try that.

ShikOfTheRa, 2018-09-11:
I have amended the source code to hopefully address this as long term solution - its in teh main source repository and will be included in next release.

Should only impact iNAv users with fixedwing and no GPS. I think there are probably not many users like this, but I am also one with no GPS!

ShikOfTheRa, 2018-09-14:
Available from within GUI as a "test" release if you wish to flash 1.9.0.1 onwards.

Closing this as I think done. Re-open - or comment if need any further action as I receive comments on closed issues too.

closed 2018-09-14 by ShikOfTheRa