ShikOfTheRa/scarab-osd #553: Where can I set screen ratio 4:3 or 16:9 in Micro Minim OSD?

GrayMagellan, 2018-10-03:
Where can I set screen ratio 4:3 or 16:9 in Micro Minim OSD? My FPV camera is 4:3 screen ratio only. My FPV monitor can show picture as 4:3 as 16:9 screen ratios. But in both cases picture is cropped on upper and lower edges of screen. One man told me to set correct screen ratio inside Micro Minim OSD (set it to 4:3 ratio on all three items of FPV path, so FPV camera screen ratio = Micro Minim OSD screen ratio = FPV monitor screen ratio), but I don't find any settings in MWOSD Configurator software.

ShikOfTheRa, 2018-10-03:
Sorry this isn't a setting or something can do in mwosd. It only adds overlay to existing signal from camera. Many lcd screens do not display first and last lines of display. Mine are same... in goggles it is fine.

GrayMagellan, 2018-10-04:
Ok, I understand. Thank you very much for this information. I am closing this issue. Thank you very much again.

closed 2018-10-04 by GrayMagellan