ShikOfTheRa/scarab-osd #555: GPS Only setup sees no GPS signal

stephenaa, 2018-10-14:
Hardware: microminimosd
Software:1.9.0.5, prebuilt and compiled from source

I cannot get any GPS data on the display, it keeps saying 0 sats.
So far I have tried with a Beitan BN 220 and a V.KEL GPS module - both communicating fin with the UBLOX u-center software.
The serial port of the OSD module is working fine for both GUI and reflashing, so I am guessing there is a software issue at play her.
Is there any known issues that may explain this?

ShikOfTheRa, 2018-10-14:
Beitian should work well set as UBLOX. 
I also have V.KEL and I have problems with anything other than 9600 NMEA mode.

So shouldn't be an issue. Double check GPS TX>OSD RX and GPSRX>OSD TX is is majority of issues. It is usually reversed connection to using USB adapter.

If no luck I will make a debug version to help troubleshooting...

Actually I have been flying 1.9.0.x a lot with prebuilt and Beitian BN180 GPS only:

https://www.youtube.com/watch?v=dbrIT_IbUAM

stephenaa, 2018-10-14:
I think I had a BAUD rate issue with the module not taken my setup to 115200, I also set the GPS to ublox though. Now it works. While trying to find the cause I looked at the GPS initialization in GPS.ino - couldn't the autobauding be used for both UBLOX and NMEA format?

Maybe I could have a try and do a pull request - unless there are other plans?

ShikOfTheRa, 2018-10-14:
That's great.

Setting ublox will enable the code that auto sets up the GPS baud rate if it can.

Unfortunately with NMEA there is no standard for setting up baud like there is with ublox. So have to set it to match GPS

stephenaa, 2018-10-14:
I might have a look when I find some time. I am somewhat working in the business...

closed 2018-10-14 by stephenaa