ShikOfTheRa/scarab-osd #556: OrangeRX cPPM output

stephenaa, 2018-10-16:
I can't seem to get a micro minim old in GPS standalone mode to accept any input from a Orange RX (R615x). Neither for menu control or just a screen switch.

Could there be some input pin confusion (Rx is connected to RSSI pad) ?

I will try to activate some debug to see if any signal is derived from the PPM signal at all.

ShikOfTheRa, 2018-10-16:
Hi - did you install via arduino  with:
//#define PPM_CONTROL               // Enables full OSD menu, screen switching, RSSI, Throttle feature, virtual current sensor, etc using a PPM signal into OSD PWM input pin (RSSI pin on basic MINIM hardware). Requires TX type to be set below.

stephenaa, 2018-10-16:
Yes, I did. However I am a bit confused about the need to set the Tx type further down in Config.h. As far as I can read, GlobalVariables.h uses either TX_GUI_CONTROL (set in Def.h when GPSOSD) OR one of the specific Tx types. 
My GUI has the GPS OSD pane grayed out, So I cannot set it there. I tried to set TX_RPTY with no success. My TX (Spektrum Dx6) sends throttle, rudder, elevator, aileron.

Does the debug funtions easily support dumping measured PPM values onto the screen?

stephenaa, 2018-10-17:
I have hooked up an oscilloscope that allows me to look at the PPM signal from the RX. I can send you some scrrenshots of that, or try to get some debug output myself. 

I noticed that the PPM signal seems to be active low.

ShikOfTheRa, 2018-10-30:
Sorry - still haven't had a chance to look at this yet. Had a fire which shut office for 2 weeks.

stephenaa, 2018-10-31:
Sorry to hear - hope you got everything sorted out.

I didn't have a change to look at it either, but I still have a plan to do so. I will keep the issue updated.

ShikOfTheRa, 2018-11-09:
Finally had time to look. I can see the problem straight away. 
Makes me think... it never worked no - one uses it!

Will resolve soon...

ShikOfTheRa, 2018-11-13:
It is resolved (hopefully!) and will be available once I have the updated GUI that will support changes. 
Will update here once available

ShikOfTheRa, 2018-11-15:
Hi, available for test.

Download latest GUI from here:

flash latest firmware (will need to tick the show test releases option)

stephenaa, 2018-11-15:
Thanks, i hope to be able to test it later today. 

Sendt fra min iPhone

ShikOfTheRa, 2018-12-07:
Closing - reply if any further issues

closed 2018-12-07 by ShikOfTheRa