ShikOfTheRa/scarab-osd #559: Arduplane RSSI either 0 or 100%

Avi8orn8, 2019-01-01:
First Github contribution, so please be nice. 

I am running a Pixracer with Arduplane and Minim OSD with MWOSD 1.9. RSSI source is set to “RSSI from Flight Controller” and the OSD indication is either 0% when my transmitter is off, or 100% when my transmitter is on. Prior to MWOSD I was using MinimOSD Extra on this hardware, which worked perfectly (good RSSI resolution). Is there something I may have missed, or is this a bug? I will be bypassing the FC and running the RSSI wire directly to the Minim OSD and using “Direct OSD connect Analog” for now.

ShikOfTheRa, 2019-01-02:
Not aware of any issues, but I have found variances in how different FC send the info so need to look into it. 
Should be set just to RSSI from FC so sounds like set up OK.

Would you be able to capture a tlog so I can look at data sent to OSD? I can play it into mine then to verify / troubleshoot.

Avi8orn8, 2019-01-02:
Hi Shikra,

Yes I can do that. I’ll try to get it to you today. Thanks!

Nathan

ShikOfTheRa, 2019-01-21:
OK I will try to emulate this. I will get a pixracer to test. Not quite sure what difference is.

ShikOfTheRa, 2019-01-25:
Hi, might have answer for you.

Short answer:
In GUI, enter 2600 as the upper calibration value. It will highlight in red as a warning it is outside of expected values. Let me know if this works. 

Long answer:
in mavlink specifications, FC is supposed to send a value of 0-100 for RSSI, 
In logs I have captured, I am seeing a value of 0-255 for the strength. So it looks like FC is not sending correct values.  
So on OSD, even at highest value calibration value you will see 100% displayed when real RSSI is down to 30%.

Here is the spec for mavlink:
https://mavlink.io/en/messages/common.html
(rssi | uint8_t | % | Receive signal strength indicator. Values: [0-100], 255: invalid/unknown.)

So what should happen is  FC should only sends a % value of 0 to 100, but it seem this is not the case.
With default RSSI upper calibration of 1023, this will give you a nice 0-100% rssi display

Avi8orn8, 2019-01-25:
Hi Shikra,

Thanks. I actually tried entering 1023 in the upper limit box and it seemed to help, but not solve the issue. It does not allow values any higher than this.

ShikOfTheRa, 2019-01-26:
Hi - if click into the cell you can manually type in 2600 into the cell. I have tried on a copter and it worked for me so hopefully it will resolve for you.

Longer term I'll get the GUI amended to support. It will likely be a good few months before next update as it will probably tie in with the next release of MWOSD.

I have updated the ardupilot / px4 install notes for now as interim.

closed 2019-02-12 by ShikOfTheRa

umeloman, 2021-03-08:
Same problem with Ardupilot when RSSI from FC, 0% or 100%. Appears only when Load Settings From File. If I flash it and configure manually - working fine