ShikOfTheRa/scarab-osd #560: Feature request: display a channel as a PWM number or a line

JeyPi, 2019-01-14:
I use a panable camera during a flight. It would be nice to be able to set a channel icon (5-12ch) visible for that function. Specially if one does not have mechanically noticeable mid positions  on the TX pan pots and the camera is stabilised.
When so the pilot have no idea what is the camera position relative to the fuselage if no part of the plane is in the view.
The Minim-extra OSD has the function displayed like this:
[image]

ShikOfTheRa, 2019-01-15:
Hi - could you share video of it in action? I haven't used minim-osd extra

The following is available if you are OK to create a custom install: (https://github.com/ShikOfTheRa/scarab-osd/wiki/Custom-installation-guide)

I will assess if this can be added as a feature from the GUI for next release. It depends upon memory available.

/********************       Headtracker support         ************************/
//#define VIRTUAL_NOSE              // Enables the use of a virtual nose for headtracker users where aircraft nose is not visible
#define HTCHANNEL   3               // RC chanel uses ch 1 - 8/16
#define HTSCALE     10              // Scaling of Pan Axis - Max of 10
#define HTLINE      11              // Row on which Headtracker info is displayed
#define HTDIRECTION +               // Reverses direction of pan action

JeyPi, 2019-01-15:
This is so cool. As if designed for me according to the order. Exactly what I was looking for. Perfect perfect perfect. Thank you.
When compiling I got this message but on the bench it works exactly to my delight:

Sketch uses 27346 bytes (89%) of program storage space. Maximum is 30720 bytes.
Global variables use 1703 bytes (83%) of dynamic memory, leaving 345 bytes for local variables. Maximum is 2048 bytes.
Low memory available, stability problems may occur.

closed 2019-01-15 by JeyPi