ShikOfTheRa/scarab-osd #561: Arduplane Altitude presentation

JeyPi, 2019-01-17:
Hi. I was already dealing with a similar problem a while ago.
This time I would say all is working only the shortcut for the altitude is something I quite do not understand.
I made a video to understand my point. But first let me explain how I grasp the shortcuts:
AGL Height above ground level - above the point where I started the flight.
ALT Height above the mean see level while the barometric pressure is 1013 Hpa.

Therefore when I start the planes autopilot I should get its ALT. (in my case 250m above the MSL)
When I start the flight, I should get the AGL ?

I do not know if that is the way it should work. The video might be more clear. 
When I start the autopilot I get 0 AGL, after a while I get -8 AGL, Than I lift the plane and I get -7 AGL
Then I launch the plane the height zeroes and I get rising ALT numbers as the plane climbs.
https://youtu.be/wGaEV0xvlx8

ShikOfTheRa, 2019-01-21:
actually most FPV OSD do not follow true aircraft rules for alt/agl/asl etc etc...

in MWOSD:

ALT: is data sent directly from flight controller
AGL is typically the same. There is a reason:

AGL is usually displaying extra digit of precision for those who want it - mainly those using  precise altitude measuring data such as LIDAR, . 

in Display tab, "High resolution altitude display cutoff" is where you can specify the point at which it changes. Default is zero, which is why you see it change. Actually now I think I need to improve code to make 0 disable it.

Back to negative values... 
This is what is sent by Arduplane and I find it varies a lot especially with GPS and barometric  drift. It is best to let it gain as many sats as possible and then arm immediately before flight. I believe arming is when it sets the altitude reference to zero. 
Maybe an arm/disarm sequence just befor flight will also will reset it.

The data displayed is what is sent from 

AGL is normally used

ShikOfTheRa, 2019-01-25:
Hi - I added following wiki page to hopefully clarify operation of altitude.

https://github.com/ShikOfTheRa/scarab-osd/wiki/Altitude-display

closed 2019-02-12 by ShikOfTheRa