ShikOfTheRa/scarab-osd #562: Roll angle display offset

shellixyz, 2019-02-28:
The roll angle is not displayed at the configured position

[image]

[image]

shellixyz, 2019-02-28:
Or maybe it is everything else that's shifted down. Notice the two lines between the vertical speed and total display in the OSD while there is only one in the GUI. Everything bellow the fourth line seems to be shifted down 1 line.

shellixyz, 2019-03-01:
I forgot to specify that I'm using GUI version 1.9 and revision bc477a1 of the firmware (latest master at the time I write this).

ShikOfTheRa, 2019-04-29:
Sorry Shellixyz I completely missed this as didn’t get notification. Development is currently being carried off github at the moment whilst work underway for a substantial new feature so I’ve been relying on those emails. I’ll have to check manually…

Yes, there is a shift. I think you are first to notice! There are approx. 50k user of MWOSD too…

So the screen layout is made up of NTSC 30 characters wide * 13 rows.
For simplicity, within GUI and primarily because lack of memory on the Atmega, the screen layouts are always in NSTC format.

When we use PAL, it simply inserts an extra blank line around lines 5 and 10. It’s a compromise…

So yes, there is a slight mismatch between GUI and real display. I would like at some point to put a PAL preview button into the GUI.

Hope that helps explain. BTW … MWOSD GUI is free for FC developers… send an email into virtualpilot website for refund...

closed 2019-05-28 by ShikOfTheRa