ShikOfTheRa/scarab-osd #563: MKX Power Distribution Board's OSD crashes with 1.9

andrejpodzimek, 2019-04-30:
I tried to flash 1.9.1.1 into [this PDB](https://www.banggood.com/MXK-Power-Distribution-Board-PDB-with-OSD-BEC-Output-5V-12V-3A-LC-Filter-for-CC3C-Naze32-F3-p-1031375.html). (It's a minimosd.) It works flawlessly with 1.8, but crashes with 1.9. After flashing 1.9 and restoring the default configuration, MWOSD talks to the GUI over the built-in usbserial on the PDB without issues. But it crashes when I boot the flight controller.

I have two instances of the same PDB. Both crash in a slightly different way. Both show the MWOSD splash screen, which looks OK and has the current version.
1. One PDB then shows the normal MWOSD screen with bright white font (no flickerihng anywhere), but stuck, i.e., the artificial horizon is not moving and nothihng is updated. Yet the FC is working (i.e., the drone can fly just fine). Just MWOSD doesn't communicate any more, i.e., Betaflight's menu gesture doesn't do anything and it remains stuck.
1. Another PDB crashes in such a way that it either makes the entire picture pitch black or it shows the OSD screen in odd flickering colors with black font instead of white. The change of getting one outcome or the other is ~50/50. It's also completely stuck, i.e., nothing gets updated.

I've tried quite a lot of stuff so far, but nothing helps:
* 2 different builds, with F4 and F7
* different Betaflight versions
* many different MWOSD configurations, both default and non-default

Flashing 1.8.1.0 and applying its default configuration **fixes the problem**. It doesn't crash any more, it shows all the data from the FC etc. However, this comes with a considerable drawback:
* the 1.8.x configurator can't flash anything; it gets stuck and doesn't download the hex files.
* the 1.9.x configurator looks like it can handle 1.8.x devices, but it can't. Its settings, default or not, yield a disastrously bogus mix of randomly shifted indicators, most of which end up outside the visible image.

So, in summary, my workaround is:
* Open the 1.9.x GUI and use it to flash 1.8.1.0. Don't write any MWOSD settings from it.
* Open the 1.8.x GUI, restore defaults, upload your favorite font etc., write your settings.

Surprisingly, I have already flashed very old `SPRACINGF3` and `BEEROTORF3` boards with built-in MWOSD and they run 1.9.1.1 perfectly fine. They don't crash.

So it's something about this particular PDB that doesn't work well with MWOSD 1.9. No clue what it is. But because I have two such PDBs, both of which work with 1.8 and crash with 1.9, it's unlikely to be just a broken specimen.

andrejpodzimek, 2019-05-01:
I have just updated the title a few times. For a short time I thought this could depend on the number of motors. (Yet another experimental hypothesis disproven.)

* Two `minimosd`s (in the PDB mentioned above) freeze in hexacopters.
* A `microminimosd` freezes instantly in my octocopter.
* The same `microminimosd` works perfectly in a quadcopter.
* Old on-board `minimosd`s works in a quadcopter (on `SPRACINGF3`).
* Old on-board `minimosd` **works in a hexacopter** (on `BEEROTORF3`).

As far as I can tell from a few hours of experimenting, the problem is independent of:
* MWOSD font. Tried large, bold, default. No effect on freezing.
* MWOSD settings. Tried minimalistic ones. Tried to eliminate current-related ones, GPS-related ones etc. No effect on freezing.
* Flight controller. It freezes at least with an `ANYFCF7` and a `BLUEJAYF4`.
* MWOSD type. The PDB freezes in a way similar to the tiny `microminimosd`.

Up until a while ago, I thought that it only froze on builds with more than 4 motors, but now I see that my `BEEROTORF3` hexacopter wors fine with 1.9.1.1.

andrejpodzimek, 2019-05-01:
Forgot to emphasize: 1.8.1.0 **works** everywhere. So it's unlikely to be just my pile of broken hardware. :grin: The workaround is to simply downgrade to 1.8.1.0 in cases where it freezes.

ShikOfTheRa, 2019-05-02:
I bet my pile of broken hardware is bigger than yours :) lol

OK, I will set up test. I suspect a bug there. Possibly memory or overflow. I had on other report for Betaflight but he is using development release for 1.10 which has some additional functionality but that may be unrelated.

ShikOfTheRa, 2019-05-14:
Hi - can you try R1.9.1.2 onwards. 
Will need to select "show test~" when flashing from GUI

I found issue. Looks like BF is sending different non real rtc date to previous and MWOSD is not handling it well. This should catch error.

closed 2019-05-28 by ShikOfTheRa