ShikOfTheRa/scarab-osd #564: List of questions

liguo-git, 2019-05-05:
1、inav2.1.0 pid  dysregulation；
2、inav2.1.0  GPS height not displayed

SPRACINGF3   microminim.inav.multirotor.hex

liguo-git, 2019-05-06:
[image]

ShikOfTheRa, 2019-05-10:
Hi,
For 1 – do you mean to add PID display support?

For 2 - 
 
[image]

liguo-git, 2019-05-14:
For 2 is small house？

ShikOfTheRa, 2019-05-14:
Hi, 

Below is screen with "Show both positions"
it is H instea dof house, but you can create your own icon if you prefer house.
 
[image]

liguo-git, 2019-05-17:
oj8k， Thanks

closed 2019-05-28 by ShikOfTheRa

liguo-git, 2019-06-07:
Map mode display type
I tried
show home position show icon ok
No other