ShikOfTheRa/scarab-osd #566: RSSI FlightOne problem

BlacScrap, 2019-05-22:
Hi, i have problem. RSSI not working in my MinimOSD.
I typed in console FlightOne - set telem_msp=4,
TX4 pad bridge to normal,
I set in my taranis rssi in 8 or 16 channel,
In MWOSD - RSSI tab set "RSSI Source" - RSSI from RC channel, "RSSI Channel" set 8 or 16.
RSSI stands still in 73%. 

I don't know why not working... Help me :(

ShikOfTheRa, 2019-05-22:
Hi - installing debug will help see channels 1-8 to see if data is coming OK from the FC:
[image]

Also, F1 is current set to use only 8 channels.. Will need to:
1 https://github.com/ShikOfTheRa/scarab-osd/wiki/Custom-installation-guide
2 in config.h enable and amend to #define TX_CHANNELS 16

In not sure how to do this, let me know and I'll creat a build you can test / install.
If this works, I'll release anew version with 16 channels enabled as default.

BlacScrap, 2019-05-22:
#define TX_CHANNELS 16 - it works!

Thank you so much! :)

ShikOfTheRa, 2019-05-22:
Thanks for feedback. I'll enable permanently and make this available as precompiled release for flashing in next couple of days.

closed 2019-05-28 by ShikOfTheRa