ShikOfTheRa/scarab-osd #567: mw_osd fails to compile

atarijookie, 2019-05-27:
Hello, 
I've cloned the mw_osd from master:
_commit 1a19f9eb (HEAD -> master, tag: 1.9.1.2, origin/master, origin/HEAD)_

I've un-comented following in _Config.h_:
#define MINIMOSD
#define GPSOSD_NMEA
#define FIXEDWING

I'm trying to compile it using Arduino 1.8.8.
Board: Arduino Pro or Arduino Mini
Processor: ATmega328P (5V, 16 MHz)

...but I'm getting lots of compilation errors:
```
...\mw_osd\GPS.ino: In function 'void gpsvario()':

GPS:1104:18: error: 'timer' was not declared in this scope

GPS:1106:5: error: 'MwVario' was not declared in this scope

GPS:1106:16: error: 'GPS_altitude' was not declared in this scope

GPS:1106:31: error: 'previousfwaltitude' was not declared in this scope

...\mw_osd\MW_OSD.ino: In function 'void setup()':

MW_OSD:148:15: error: 'checkEEPROM' was not declared in this scope

   checkEEPROM();

               ^

MW_OSD:149:14: error: 'readEEPROM' was not declared in this scope

   readEEPROM();

              ^

MW_OSD:166:16: error: 'MAX7456Setup' was not declared in this scope

   MAX7456Setup();

                ^

MW_OSD:180:18: error: 'setMspRequests' was not declared in this scope

   setMspRequests();

                  ^

...\mw_osd\MW_OSD.ino: In function 'void loop()':

MW_OSD:278:15: error: 'resetFunc' was not declared in this scope

     resetFunc();
```

What's wrong here? Something with the source and Arduino IDE of my version not playing along? :) Is there an easy fix for this? 

Because if not, I was thinking of generating header files for missing function declarations, or cutting down the sources to my minimum need, but I would rather do as minimal changes as possible, as I'm considering adding a feature here ;)

atarijookie, 2019-05-27:
Fails to compile on Arduino 1.5.5-r2 - can't initialize non-static members of struct: GUI_active=INTRO_DELAY;

Compiles fine on Arduino 1.6.0

ShikOfTheRa, 2019-05-28:
1.6.5 is used for development. This is what I current recommend. We use this as there are some advantages for memory testing.

Compiled released are created using latest gcc on travis platformio.

I am aware there can be issues with other versions. I will try to resolve those shortly. One of those issues I am aware of but hasn't been raised. It is purely an issue of order in which the compiler processes. It should be ameneded.

atarijookie, 2019-05-28:
Okay, thanks, I'm using 1.6.0 and I might switch to that 1.6.5... I guess we can close this and you can take a note to fix this in the future ;) (and possibly update / add the recommended Arduino version in the wiki here, if it already isn't there and I've just missed it?)

closed 2019-05-28 by atarijookie