ShikOfTheRa/scarab-osd #569: ver 1.9.1.1 strange characters minimosd pixhawk

66nola66, 2019-06-23:
I am running ver 1.9.1.1 on minimosd and pixhawk. I am in correct telem port and everything is responding correctly but I have strange characters showing up for the altificial horiz. and the alt. and speed side bars(these look like padlock with a G in the middle). I am running BOLD fonts and I am powering both sides of the minim from the FC 5vdc. The artificial horiz. actually has random letters making up the bars and center marker. Any help would be greatly appreciated. Thanks!

error414, 2019-06-23:
Hi 66nola66,

can you attach picture how it looks like? I tried find char which would looks like padlock wig G but without success. 

Did you try re-upload new font?

66nola66, 2019-06-23:
[image]
[image]
[image]
[image]

I havent tried to reload new font but will do that next. I did hear that when loading a file it needs to be done with both sides of the minim powered which I didnt do originally but have since done with similar results. Here are the PICs you requested. AND THANK YOU!!

error414, 2019-06-23:
I am sure that load new font is the key :).

66nola66, 2019-06-24:
That did it!!!! THANK YOU!!! EXCELLENT SUPPORT!!!!

closed 2019-06-24 by error414