ShikOfTheRa/scarab-osd #573: KISS 1.3RC39b : size of KISSserialBuffer too small

xaviergriffon, 2019-08-20:
The size of the "KISSserialBuffer" array is too small with KISS 1.3RC39b. In my tests, the frame length was 170.
I would advise using a size of 256 (KISSFRAMELENGTH) as KISS OSD does.

I will test flights tomorrow to validate that this is the only issue ;)

xaviergriffon, 2019-08-21:
I tried other solution with a simple "if" when we stack buffer.
```
  else if (c_state == KISS_HEADER_SIZE) {
    if (Kvar.index < KISSFRAMELENGTH) {
      KISSserialBuffer[Kvar.index] = c;
    }
```

Currently, the last index (for Mah) is 148. This solution allows to use a smaller buffer (149).

ShikOfTheRa, 2019-08-21:
It has grown over time :) Thanks for feedback. Not sure if many use MWOSD for KISS. I get that it has its own but would like to see it more widely utilised. 
KISS is awesome and rated #1 by the MWOSD GUI dev :) 

Memory is in very short supply without some time consuming optimisation, but if any additional functionality needed to match what KISSOSD offers happy to look at it. I think dev on that has stopped?

So suggest:

in globalvariable.h change:
#define KISSFRAMELENGTH 200 // max frame size

in serial.ino add:
#elif defined KISS
  #define SERIALBUFFERSIZE 65
#else

increase KISS data buffer significantly.
Reduces MSP serial buffer (only used for GUI) by same amount

Net change = zero so should be no impact on functionality.

fedorcomander, 2019-08-21:
Present for you...

	// link stats, crossfire only for now

        	kissProtocolWriteByte(RXLinkStats.upRSSI1, &dst);
        	kissProtocolWriteByte(RXLinkStats.upRSSI2, &dst);
        	kissProtocolWriteByte(RXLinkStats.upLQ, &dst);
        	kissProtocolWriteByte(RXLinkStats.upSNR, &dst);
        	kissProtocolWriteByte(RXLinkStats.upAntenna, &dst);
        	kissProtocolWriteByte(RXLinkStats.rfMode, &dst);
        	kissProtocolWriteByte(RXLinkStats.upTXPower, &dst);
        	kissProtocolWriteByte(RXLinkStats.downRSSI, &dst);
        	kissProtocolWriteByte(RXLinkStats.downLQ, &dst);
        	kissProtocolWriteByte(RXLinkStats.downSNR, &dst);

You also can get gps data (via differenc command), lmk if interested.

Alex

xaviergriffon, 2019-08-21:
I created a PR (#574 ) in which I left comments and definitions that I used during the analysis.
No problem to delete them if you want or to make the simplest with the buffer at 200 and without if.

ShikOfTheRa, 2019-08-23:
@ Alex
 
> You also can get gps data (via difference command), lmk if interested.
> Alex

Definitely interested in the GPS data!

I don't use crossfire so that will be a challenge for me..

ShikOfTheRa, 2019-08-23:
@Zavier - its added. Many thanks for that

fedorcomander, 2019-08-23:
KISS_PROTOCOL_GET_GPS  		 =  0x54,

  case KISS_PROTOCOL_GET_GPS:
            kissProtocolWriteInt32((int32_t) (gps.latitude * 10000000.0f), &dst);
            kissProtocolWriteInt32((int32_t) (gps.longitude * 10000000.0f), &dst);
            kissProtocolWriteWord((uint16_t) (gps.speed * 100.0f), &dst);
            kissProtocolWriteWord((uint16_t) (gps.course * 100.0f), &dst);
            kissProtocolWriteWord((int16_t) gps.altitude, &dst);
            kissProtocolWriteByte((uint8_t) ((gps.fix << 7) | (gps.satellites & 0x7f)), &dst);
            resp_len = (uint16_t)(dst - &response[2]);
            break;

ShikOfTheRa, 2019-08-23:
Brilliant - I will look to implement next week.

fedorcomander, 2019-08-23:
Lmk if u need any help. buzz me for gps capable firmware if u need.

ShikOfTheRa, 2019-08-23:
Is firmware available for V1 board? Guess  not enough resources?

fedorcomander, 2019-08-23:
sure it is... plenty of resources yet on v1...

fedorcomander, 2019-08-23:
its not taking that much cpu, only adaptive filter is heavy, but hey, you do need to crunch that numbers ;)

ShikOfTheRa, 2019-08-23:
Fantastic - can you drop a mail to gary  (hat) flypix.co.uk and we can take this offline. It is cool development for KISS. seeing  a lot more guys doing longer range and want GPS

error414, 2019-08-23:
It is great news, may I ask you for FW for KISS V1, does it need new configurator too? My email error414@error414.com . I would like to test it, if it is possible.

fedorcomander, 2019-08-23:
Yeah, i am hooked on LR now, so need to have GPS kiss goodies for myself ;)

ShikOfTheRa, 2019-08-23:
Alex - error414 is the GUI dev..

fedorcomander, 2019-08-23:
mailed you, you can distribute along your devs. please dont share, its my branch, not yet official.

xaviergriffon, 2019-08-24:
@ShikOfTheRa: Thank you for integrating the correction.
If you need any help for tests or implementation, do not hesitate to contact me by mail griffon.xav (hat) free.fr

For me, this issue can be closed.

fedorcomander, 2019-08-27:
> Thanks both
>
> Xavier - I'm going to keep open a little longer as a reminder for me to make sure I add in the GPS support.
>
> Alex - yes agree no sharing outside the 2 core devs

okidoki

---
Cheers,
Alex

ShikOfTheRa, 2019-08-27:
Thanks both 

Xavier - I'm going to keep open a little longer as a reminder for me to make sure I add in the GPS support. 

Alex - yes agree no sharing outside the 2 core devs

xaviergriffon, 2019-09-06:
> KISS_PROTOCOL_GET_GPS = 0x54,
>
> case KISS_PROTOCOL_GET_GPS:
> kissProtocolWriteInt32((int32_t) (gps.latitude * 10000000.0f), &dst);
> kissProtocolWriteInt32((int32_t) (gps.longitude * 10000000.0f), [...]

@fedorcomander, does KISS_PROTOCOL_GET_GPS instruction work in rc40i ?
I tried (your sharing on RCGroups :)), but I have no answer, no more instructions work after for a very long time. I tried via the serial port and directly in USB but it's the same.

fedorcomander, 2019-09-06:
yeah, i had to add this command to “backward compatibility” commands - single byters…. ;) fixed in 40j

xaviergriffon, 2019-09-15:
Thanks @fedorcomander and sorry to respond so late ...
Is it important to switch all requests to the format: "Command" "Size" "CRC" ?
I tried for the GPS and it works better :)

ShikOfTheRa, 2020-03-27:
@ xaviergriffon - All the KISS items are released in 1.9.1.5 

1.9.1.5 Summary of changes :
OSD - Support for KISS GPS
OSD - Support for KISS PID menu support
OSD - Support for KISS Filter menu
OSD - Support for KISS VTX menu
OSD - Support for ARDUSUB
OSD - Support for MAVLINK dual GPS / Baro altitude display 
OSD - Support for MAVLINK high resolution altitude
OSD - Significantly improved Mavlink messages. Full length and human readable
OSD - Different indicator icons for GPS, Baro and high resolution altitude values 
OSD - Bugfix for iNav airspeed display
OSD - Bugfix for altitude display. GPS can be displayed independently of Baro altitude 
OSD - Developer debug memory display functioning for newer compiler versions

ShikOfTheRa, 2020-04-23:
Closing. I think all KISS is working OK from feedback
Also have support for Steele PDB.

Cool - many thanks Xavier

Re-open if need

closed 2020-04-24 by ShikOfTheRa