ShikOfTheRa/scarab-osd #579: Ardupilot + MWOSD incompatibility with voltage over 30V (12S LiPo)

umeloman, 2019-09-17:
Next setup: Pixhawk Cube Arducopter 12S LiPo 48V, MAUCH Current sensor+ BEC, in Mission Planner everything is correct, voltage and current.
 Voltage from FC, differrent firmwares MWOSD 1.7 or 1.8 or 1.9 show voltage just like it recognized as 1S lipo (f.e. 4.5V). So it is only 6S compatible?

ShikOfTheRa, 2019-09-18:
Hi - yes it is only 6s. Wow. 12s - it is first have been asked for > 6s in 5 years!

12s. Do you have link? wold love to see that beast.

I can update, but am on holiday so will be about 2 weeks

umeloman, 2019-09-20:
Hello! Thanks for reply! I've tried to change various settings in ino
firmware files but has no luck.
This is big hexacopter Gaia160
http://piccy.info/view3/13401242/0e6097704857ee7f37131d55157a45eb/orig/

ShikOfTheRa, 2019-09-23:
I am back from holliday and can take a look at this in next couple of days.

That is a beast!

ShikOfTheRa, 2019-09-25:
Hi, have something available to test for >6s for Mavlink users

It is in banch here:
https://github.com/ShikOfTheRa/scarab-osd/tree/12s

Download zip available here:
https://github.com/ShikOfTheRa/scarab-osd/archive/12s.zip

I am unable to test, but if you are able to verify it is OK I will add to the master and put it up as a pre-compiled release.

Only one caveat:
At the moment, the alarms must be set to use “Autodetect cell count and alarm value” this is the default setting.

ShikOfTheRa, 2019-09-26:
Hello - tested to 80V with Analogue  and seems OK. 
Hopefully APM/PX4 is working. Should be good for up to 300v

https://youtu.be/MfRk_7FZPG4

umeloman, 2019-09-26:
Hello! It's beautiful! Thank you for this! I'll try it soon and let you know

umeloman, 2019-11-06:
pixhawk cube fc, 
#define MICROMINIMOSD
#define APM
#define BAUDRATE 57600
“Autodetect cell count and alarm value”

here what I got https://youtu.be/PX_hhyj-2l8

ShikOfTheRa, 2019-11-06:
OK - I tested with direct connect voltage and it looks OK
. I'll try to simulate tomorrow with 12S from APM.
Are you able to capture a tlog? I am not quite sure how you do this, but it means I can replay your data into OSD to simulate exactly.

umeloman, 2019-11-06:
https://drive.google.com/open?id=0B4DIAsxlndDVTGJMb2pEUVF6X3lPZUJIZWpUdU1hVmtsRF9z  my tlog

ShikOfTheRa, 2019-11-07:
Thank you for the files – it is really helpful.

I missed one other element which causes issues with voltages above 32v. It only showed up with mavlink data so I did not pick it up with testing using direct connection. 

Here is video showing your data played into OSD for testing. If you have not seen this functionality – on the GUI, select “Load LOG from disk” then can play and adjust speed. It is not timed to be exactly as on real aircraft as it is mainly for troubleshooting.

https://youtu.be/Xyz71-SlXGs?t=249

Can you retry loading 1.9.1.4 with todays updated release? I haven’t changed the version number yet as its not released.

Sorry for the trouble. It is unfortunately the pain of being at the leading edge and pushing the limits.
It is still amazing to see that Power on the OSD – 12S with over 50A. Absolutely unreal…

ShikOfTheRa, 2019-11-20:
Hopefully this is now resolved. Available in release 1.9.1.4 out today so closing.

closed 2019-11-20 by ShikOfTheRa