ShikOfTheRa/scarab-osd #582: AHI Horizon Roll reversed

nightfury24, 2019-10-05:
Hello,

As per thread "Inverted AHI pitch #413". I have make the changes in config.h but it does not reflect any changes to AHI horizon functionality. When plane roll right AHI horizon tilt left(vice versa). 
Using:
1) Mwosd 1.9.3 with GUI configurator + make changes in config.h used Arduino ( as per mwosd guide).
2) Pixhawk 2.4.8 latest ardupilot.
3) Aeromax OSD
All other telemetry working perfectly.

Horizon shows correctly in Ardupilot via USB. I understand thats how it supposed to work as we are looking horizon not the actual plane but it very... hard to maintain the relation between brain and horizon on the screen. What so ever, many times brain reflect/respond to horizon tilt and plane goes different way till you suddenly remember to follow other way around.

Any help would be highly appreciated.

nightfury24, 2019-10-05:
RESOLVED as described in thread "OSD artificial Horizon is going in the wrong direction on roll #2313"
-----------------
While rolling the aircraft the artificial horizon is moving the same way as the aircraft not the other direction as it should be. (artificial horizon should move with real horizon not in the other direction)
I corrected this by changing
int y = (rollAngle * x) / 64; to
int y = (-rollAngle * x) / 64;
in osd.c line 316 in the current master branch.
----------------------------------------------------------------

Other settings, advised in other threads did not resolved the issue. eg: in thread " Inverted AHI pitch #413"

ShikOfTheRa, 2019-10-07:
Hi - I enabled the following to test and it worked OK for me.:

#define REVERSE_AHI_ROLL          // Reverse pitch / roll direction of AHI - for DJI / Eastern bloc OSD users.

closed 2019-10-10 by ShikOfTheRa