ShikOfTheRa/scarab-osd #587: source doesn't compile with target minimosd, RTFQV1 and px4

georgemulder, 2019-12-24:
Hi,

Uncommenting this in config.h
#define MINIMOSD                  // Uncomment this if using standard MINIMOSD hardware (default)
#define RTFQV1     
#define PX4    

Results in:
Def.h:1039:24: error: #elif with no expression

 #elif ALTERNATEDIVIDERS

ShikOfTheRa, 2019-12-28:
Hi - thanks for the pull request.
Guess this is very rarely used . Amazed it has not come up before.

georgemulder, 2019-12-30:
Hi,
I have a minimosd that ships with the chinese pixhawk and I follow this out-of-date guide: http://dronesdecarreras.com/tutorial-como-configurar-un-minimosd-paso-a-paso/ That says:

```
- We comment on the line that puts MINIMOSD with two bars (//): «// # define MINIMOSD» and uncomment the WITESPYV1 line by deleting the two bars (//): «#define WITESPYV1» for MinimOSD or «#define WITESPYMICRO» for Micro MinimOSD.
```
In the actual src there isn't "WITESPYV1" but RTFQV1 have references in the comments so this is the reason i have done that config. Maybe i do it wrong.. I have not tested it yet.

georgemulder, 2019-12-30:
I have read the guide again and i realice I did something wrong... I have no need for use RTFQV1 so the destiny found the bug :)

ShikOfTheRa, 2019-12-30:
Hi,
I believe Witespy became RTFQ. 

That is funny - but it is good to remove bugs so I am happy!
Do you have code for new version? If not, please send email to gary (at) flypix.co.uk so I can send to you

closed 2019-12-31 by ShikOfTheRa