ShikOfTheRa/scarab-osd #592: Setting serial speed

thetiana, 2020-02-04:
How i can set speed to communicate with the FC

error414, 2020-02-04:
If you want change baud rate you have to compile FW, you can use baud rates from list, see  https://github.com/ShikOfTheRa/scarab-osd/blob/master/MW_OSD/Config.h#L174

Compile FW is easy if you use arduino ide. 

Why do you want change baud rate?

thetiana, 2020-02-04:
I am running out of UART ports on my FC and i want to connect OSD to RX line on the MAVlink Radio module.

error414, 2020-02-04:
ok, so change baud rate in config file which I posted above and then follow this guide https://github.com/ShikOfTheRa/scarab-osd/wiki/Custom-installation-guide

thetiana, 2020-02-04:
Is there exist some light version of this code, i am running this on 328p after compile the code its show use 80%. From my experience with 328p this is too much code for this tinny uC. More than 50% is risky for this uC. Unfortunately i fucked up my OSD Board, after upload changed code, symbols jumping all over the screen data appear and disappear on the random places on the screen. After roll back per-compiled firmware there no icons anymore, the artificial horizon is made of alphabet characters. I do character update with all possible character sets but no success.

ShikOfTheRa, 2020-02-05:
No lite version. You can remove functionality out of existing if you want. Some of it is identified in config.h

Flash and eeprom memory memory can be 99.9% utilised and it does not matter. 

What does matter is free RAM. If this runs out then there usually is unpredictable results.And compiler cannot accurately measure this. I check this on core precompiled builds to ensure there is a little free after running all the core functionality.  Using the debug screen you can monitor program ram to determine if this is an issue:
//#define DEBUG 4                   // Enable/disable option to display OSD debug values. 

For fonts
If you use a standard minim board, that uses 12v on cam/vtx pins, you will need to power this output stage to be able to flash fonts.

closed 2020-02-18 by ShikOfTheRa