ShikOfTheRa/scarab-osd #593: Can't get in to the menu (Raceflight Millivolt v1)

artemaminov, 2020-03-03:
THR MID + YAW RIGHT + PITCH FULL (UP)
Voltage and Arm status are correct

FC Version - 1.2.94.193 Stabel
FC Hardware - Flight One Revolt (Millivolt v1)

MWOSD 1.9.1.4 FW
Millivolt UART3 TX->RX and RX->TX connected to micro minimosd
set set telem_msp=3

ShikOfTheRa, 2020-03-03:
Hi,

To enter menu the following need to be correct.
Disarmed
Pitch >   1850
Yaw >  1850
Throttle < 1150

They also need to be received in correct order from the flight controller. I am not sure how well Flight one follows the MSP protocol.

To verify,
Flash 1.9.1.4, utilities, debug, msp115k

On screen RC values received will be shown and first four shoul dbe in order 
Roll,
Pitch,
Yaw,
Throttle.

You can see if values are received, order is correct and meet the criteria to access menu.

artemaminov, 2020-03-03:
No RC data coming. Millivolt v1 is not supported?

ShikOfTheRa, 2020-03-03:
Hi I suspect FlightOne is not sending the data. Thought that might be the case actually.  I have had a look at their documentation and it doesnt say much about MSP serial. Recommend raising it on their support forums as a request.

closed 2020-03-18 by ShikOfTheRa