ShikOfTheRa/scarab-osd #599: Can Mr Steele PDB support be added?

sevet, 2020-04-17:
Can you please add support for Steele PDB?

ShikOfTheRa, 2020-04-17:
I suspect it will. I can give you some bullet point instructions if you would like to try

sevet, 2020-04-17:
Sure, im in, 
im not a coder, but have some expertise ;) 
Tried to look at  wolf kiss osd to see the difference in the code but lost it....

ShikOfTheRa, 2020-04-17:
Do you have MWOSD GUI configurator? if not send an email requesting it to gary <hat> flypix.co.uk

Lets see if can flash it from the GUI first. It looks like it uses same layout as minim osd except maybe led will need a tweak.

Once get GUI, select version 1.9.1.5 and try flash kiss multirotor gps option.
If flashes OK, will then need to flash fonts.
Then its time to try it to see if it works

sevet, 2020-04-17:
Can't seem to flash. i got the com port and flashing it doesn't go pass waiting for bootloader reset, 
Tried all speeds, it needs both USB and battery connected to flash
Also disconnected it from the KissFC so it won't interfere but nothing :(

ShikOfTheRa, 2020-04-17:
Ok then can either try flashing precompiled hex file in same way that would with Kiss OSD. This is available from github on the releases link. Latest version is only one with Kiss GPS version.

Or could download the code, edit config.sys file (only need to select kiss gps) , select right board in arduino ( guess select uno board).

As FYI, arduino always programs at same speed. No need to select.
In operation OSD and kiss run at 115k

ShikOfTheRa, 2020-04-18:
OK, had a bit more of a look. We flash at 56k to support bootloader on standard minim board. Steele looks like it has a 115k bootloader. So I think MWOSD GUI will never flash unless we add an option for that.

I think maybe option 1:
•	Download precompiled kiss gps build  from
•	https://github.com/ShikOfTheRa/scarab-osd/releases/download/1.9.1.5/minim.kissgps.multirotor.hex
•	Download xloader or use the KISS OSD version
•	Select Device =  "Uno (ATMEGA328)" in Xloader
•	Flash the hex file you downloaded
•	Load MWOSD GUI configurator
•	Upload MWOSD fonts
•	Configure screen layouts as you want

or option 2:
•	Download “source code” from 
•	https://github.com/ShikOfTheRa/scarab-osd/archive/1.9.1.5.zip
•	Select "Arduino/Genuino Uno" under Tools->Board in Arduino
•	Select correct com port  and upload
•	Load MWOSD GUI configurator
•	Upload MWOSD fonts
•	Configure screen layouts as you want

Once flashed, I think you will be good to run GUI configurator. I am sure this will work once flashed.

sevet, 2020-04-18:
Thanks,

Already uploaded the precompiled hex with the xloader, configured and
uploaded fonts,
I'm not seeing anything on the OSD,
Should i see something even if the OSD and Kiss board are not communicating
(they should but i'm not certain)

ShikOfTheRa, 2020-04-18:
I had  a look at KISSOSD code and it looks like the wolfPDB uses different hardware pins to control Max IC

I have created a test branch with modified pins  (WolfPDB) to try out changes:
https://github.com/ShikOfTheRa/scarab-osd/archive/WolfPDB.zip

sevet, 2020-04-19:
Hi, I saw the precompiled hex were in 1.7.1 directory and didn't have
kissgps so i used arduino to compile.
Couldn't compile KISSGPS, Compiled the regular KISS, and it is working,
fonts uploaded and the OSD gets data from the Kiss!!
Great!! the GUI btw says hardware: unknown.

[image: image.png]

ShikOfTheRa, 2020-04-19:
Ok, that’s great. I can finalise and implement properly now we know it works. I’ll tidy it up in next couple of days and add into the master.

Do you have GPS? It should have compiled, but I’ll check this out tomorrow.

Hardware unknown is OK. GUI doesn’t know about wolf pdb yet!

I’ll do some proper instructions too

sevet, 2020-04-19:
Great, thanks.

Yes, I have GPS.
It didnt compile on the error:
GPS_reset_home_position was not declared in this scope.
(Sent screenshot on last replay)
Also. I needed to compile on arduino with: "Arduino/Genuino Uno"
Not pro mini

ShikOfTheRa, 2020-04-19:
mmm. It compiles OK for me with KISSGPS and WOLFPDB enabled. Board set to Uno too.

What version Arduino do you use? I am using a quite old one - 1.6.5 for some compatibility checks that I probably don't need any more. 
I'll try on latest.

Will tr to get all finished and tidied up tomorrow.

sevet, 2020-04-19:
I used arduino 1.8.7 or something>1.8

error414, 2020-04-19:
Hi Sevet, 

can you please try it? It works in my case(compiling), Arduino 1.8.9
[error414-osd-WolfPDB.zip](https://github.com/ShikOfTheRa/scarab-osd/files/4498912/error414-osd-WolfPDB.zip)

sevet, 2020-04-19:
This one compiles, on 1.8.8 (different computer, did also try the
previous and it didn't compile with same error)
I didn't test though to see that it works.

error414, 2020-04-19:
> Komu: "ShikOfTheRa/scarab-osd" <scarab-osd@noreply.github.com>
> <comment@noreply.github.com>
> Odesláno: 19.04.2020 16:32:01
> added? (#599)

Hi Gary,

just info, it's not problem to select for each target different flash 
speed. It's already implemented in STK500 library.  It's only about 
change one param.

Petr
------ Původní zpráva ------

Hawker2007, 2020-04-19:
Hi All =), first of all thank you for your amazing work.  error414-osd-WolfPDB.zip compiled (in 1.8.10) and uploaded to SteelPDB, OSD works, will test gps tomorrow.

ShikOfTheRa, 2020-04-20:
Merged the development into the main code and made available as a test release as well:
https://github.com/ShikOfTheRa/scarab-osd/releases

Added XLOADER files for installing:
https://github.com/ShikOfTheRa/scarab-osd/tree/master/OTHER/STEELEPDB_KISS

Created an instruction guide:
https://github.com/ShikOfTheRa/scarab-osd/wiki/Steele-PDB-for-KISS

Let us know how you get on. I would love to see more people using MWOSD for Kiss. Especially as it looks like MISS OSD is no longer being developed
A few do, but I think it is largely unknown.

sevet, 2020-04-20:
I tested this today error414 and it gives me starnge charecters:
[image]
[image]
I used bold fonts
Trying the main code now

sevet, 2020-04-20:
Still bad charecters in osd on 1.9.6 precompiled steelepdb.kiss.gps.hex using large fonts (reuploaded)
(says ImpulseRC hardware on gui now)

ShikOfTheRa, 2020-04-21:
Maybe this...
"For some of the original minim boards, the output stage (max chip) MUST be powered when uploading font. How depends upon your board / configuration"

I same note about power must be applied to the board when programming?

**MWOSD FAQ 14 I get garbage characters and / or font upload doesn't work.**

Sounds like fonts need to be re-uploaded - but note:

- For some of the original minim boards, the output stage (max chip) MUST be powered when uploading font. How depends upon your board / configuration
- A programming voltage of 4.8v or higher is required.
- Note some programmers have 3.3v/5v selection text swapped. Verify with a meter
- Some stacked boards with USB connector also require vbat connected which powers the MAX chip and enables font upload
- Programming with FC passthrough is not always successful.

sevet, 2020-04-22:
For steelePDB configuration and flashing you have to have it powered on anyways. (and also disconnect the data from the KISS FC)
I checked again and it seems i just can't upload fonts, tried rotating default/bold/large and the fonts always stay the same.

sevet, 2020-04-22:
I tried compiling 1.9.1.6 and it only compiled for KISSGPS with Arduino 1.6.5 otherwise i get that GPS declaration error, after compiling the using gui to upload fonts i still have bad characters like the previous picture (seen directly in this thread on github) 

Tried to use "define LOADFONT_LARGE" but it didn't finish uploading for a few minutes so i stopped it.

ShikOfTheRa, 2020-04-23:
I'll take the compile issue as a separate problem...

For the fonts:
I think I made an error with one of the pins used by Steele pdb.  
I have updated the main reposirory, but probably quicker to edit directly
Line 1036 in def.h should read:
 # define MAX7456RESET      9 // RESET

ShikOfTheRa, 2020-04-23:
Just to add - sorry for teh pain - its teh joy of being at teh leading edge - and us not having teh hardware!

We'll get there...

ShikOfTheRa, 2020-04-23:
Hopefully compile issue also resolved in latest master repository.

sevet, 2020-04-24:
Great! :)
Everything works perfectly - compiles on latest arduino and uploading fonts was successful :)

ShikOfTheRa, 2020-04-24:
Excellent. We go there in the end!

ShikOfTheRa, 2020-04-27:
OK - closing. Re-open if any issues arise :)
Many thanks for your time to test. Hopefully it will help others out there too.

If you get a flight video, would appreciate one up in the Kiss thread - I don't think many are aware of MWOSD.

closed 2020-04-27 by ShikOfTheRa

Machtap, 2022-06-20:
I know this is weird but would it be possible to compile a version that targets Steele PDB hardware but accepts and displays betaflight telemetry data?

ShikOfTheRa, 2022-06-20:
I think it can support that now - have you tried it?

#define STEELEPDB