ShikOfTheRa/scarab-osd #601: Avarage battery cell display and warning by avarage cell

sevet, 2020-04-20:
Can the average battery cell display please be added?
Also if the battery warning will be by the average cell it will make to possible to get a more general setup for the OSD.

ShikOfTheRa, 2020-04-20:
Yes - will add average cell voltage. 
It will have to be selected in config.h rather than GUI - at least for now.
Battery warning is by cell alarm value by default. It is in GUI main voltage tab. It is set to 3.4 by default.

I''l update when completed

sevet, 2020-04-20:
I like to have both average and actual :)
Will it be possible in the future?

ShikOfTheRa, 2020-04-20:
See what can do. Definitely if happy to use arduino to enable

sevet, 2020-04-23:
> Yes - will add average cell voltage.
> It will have to be selected in config.h rather than GUI - at least for now.
> Battery warning is by cell alarm value by default. It is in GUI main voltage tab. It is set to 3.4 by default.
> 
> I''l update when completed

I thought the 3.4 default is just for the autodetect, what if i want my alarm to 3.5 or 3.6?

ShikOfTheRa, 2020-04-23:
It's the alarm value and can set at 3.5 if you prefer. 

The values used to autodetect cell count is hardcoded, but can be changed in config.h:
/********************       Voltage Warning Settings         ************************/
#define CELL_VOLTS_MAX 42           // Specify the max normal LIPO cell voltage. Used for auto cell count determination and battery guage icon

ShikOfTheRa, 2020-04-23:
OK, I think I have a solution for you. It works on the bench OK
 
In master repository is new value in config.h
//#define SHOW_CELL_VOLTAGE         // Show average cell voltage for main battery instead of video / bat 2 voltage. 

This r-epurposes the video voltage field to show single cell average voltage.

So you can have full, voltage, cell voltage or both and move them independently around the screen

sevet, 2020-04-24:
Thanks, installed and checked, it works perfect! :)

ShikOfTheRa, 2020-04-24:
Thats great. maybe in future release I can look to make available as a default option.

ShikOfTheRa, 2020-04-27:
Closing. Re-open if needed.

And again - thanks for time taken to test.

closed 2020-04-27 by ShikOfTheRa