ShikOfTheRa/scarab-osd #602: Can VTX power and/or Channel information be added?

sevet, 2020-04-24:
Can you please add VTX power and channel? they need to be separate options (I don't really use channel, just power, but many use  it)

ShikOfTheRa, 2020-04-24:
I will have to think about approach for this. Something we can do within constraints of memory. 
Leave with me a couple of days

ShikOfTheRa, 2020-04-29:
Hi, I have implemented this and uploaded to the master repository, but have no way to test. 

When using kiss, the GUI "gimbal" field can be used to display VTX power and channel.
In config.h you can choose VTX, Power or both (default).
[image]

closed 2020-05-13 by ShikOfTheRa